Clon1998/mobileraker #447: After the iOS device is updated to 2.8.3, it cannot launch the APP normally. will get stuck on the splash screen.

zhengzhengju, 2024-12-19:
Hello！My iPhone 11 Pro and iPad devices, after updating to 2.8.3, can't start the APP normally. will get stuck on the splash screen. If you cut off the device network, you can start to enter the APP! But without the network, it is obvious that you can't use the APP normally.
And if you enter the APP, you can't save the settings normally after setting the APP language!

Regions of use: China
Use the network: mobile operator and home Wifi

Clon1998, 2024-12-22:
Hey @zhengzhengju 
can you do me a favor and upload the latest app log files (App Settings -> Scroll to the bottom -> Debug Logs) after you :
1. Started the app with wifi/cellular on and got stuck (Wait 10-30sec please) and 
2. Force close, Disable wifi/cellular
3. Started the app again (No wifi/cellular)

On my end I will try to use a VPN to emulate a Chinese connection to see what the GFW is blocking here :)

zhengzhengju, 2024-12-24:
[mobileraker_2024-12-24T20-20-20.log](https://github.com/user-attachments/files/18239490/mobileraker_2024-12-24T20-20-20.log)
[mobileraker_2024-12-24T20-18-47.log](https://github.com/user-attachments/files/18239491/mobileraker_2024-12-24T20-18-47.log)
[mobileraker_2024-12-24T20-18-54.log](https://github.com/user-attachments/files/18239492/mobileraker_2024-12-24T20-18-54.log)

zhengzhengju, 2024-12-24:
@Clon1998 
Strangely, this issue does not exist in version 2.8.2! This is terrible, all users in China are unable to use it. Can 2.8.2 be temporarily downgraded and put on the store for our use?

Clon1998, 2024-12-24:
> @Clon1998 
> 
> Strangely, this issue does not exist in version 2.8.2! This is terrible, all users in China are unable to use it. Can 2.8.2 be temporarily downgraded and put on the store for our use?

Downgrading for a specific region sadly does not work.
I will look into this today.

Clon1998, 2024-12-24:
Hey @zhengzhengju,
with your log, I think I identified "RevenueCat" as the main problem that causes the app to get stuck.
I made a few code changes to hopefully fix the startup error.

I just triggered a new beta build via TestFlight (v2.8.4), which should include the fix.

Can you please report back if the issue is resolved? If not, please reupload the three log files with the latest TestFlight version, as I am unable to find a VPN that would allow me to tunnel into the public Chinese network, which is actually "inside" the GFW, and therefore allows me to test the app as if I were in China. If you have any suggestions on how I could test this, I am open to suggestions :).

To get the latest version you can join the beta here:
https://testflight.apple.com/join/h9JwuyR0

zhengzhengju, 2024-12-24:
@Clon1998 
Thank you very much for your work! At present, the beta version of 2.8.4 has not been updated. I will first provide you with the startup screen recording of the app (this is the startup screen recording of 2.8.3). After the 2.8.4 beta version is updated, I will test it as soon as possible and give feedback here.

zhengzhengju, 2024-12-24:
Uploading copy_B7B8DB83-6C42-4D94-AD78-DD11D697CB7A.mov…

zhengzhengju, 2024-12-24:
https://github.com/user-attachments/assets/7fec0b9e-3d6e-4d43-8d51-de466373a9f4

Clon1998, 2024-12-25:
> 
> 
> https://github.com/user-attachments/assets/7fec0b9e-3d6e-4d43-8d51-de466373a9f4
> 
> 
> 
> 

Thanks, 
the recording matches with the log entries.
Please report back if the beta on TestFlight works

zhengzhengju, 2024-12-25:
> > 副本_B7B8DB83-6C42-4D94-AD78-DD11D697CB7A.mov
> 
> 谢谢，记录与日志条目匹配。如果TestFlight上的测试版有效，请汇报

Up to now, testflight has not updated 2.8.4.
[image]

zhengzhengju, 2024-12-26:
Hi @Clon1998 

Up to now, testflight has not updated 2.8.4.

[image]

Clon1998, 2024-12-26:
Hey,
The beta is still awaiting approval. Due to Xmas holidays the app validation takes longer than usual.

zhengzhengju, 2024-12-26:
@Clon1998 All right! Thank you very much for your hard work!

Clon1998, 2024-12-28:
@zhengzhengju 
Seems like apple is seriously slow during the holidays. I hope there will be an approval today.
I can't even publish a hotfix as these are also reliant on apples approval process.

zhengzhengju, 2024-12-28:
@Clon1998 
I'm trying to use v2.8.2 in TestFlight, and the problem still exists. You can't access the App if you connect to the Internet.

Clon1998, 2024-12-28:
> @Clon1998 
> 
> I'm trying to use v2.8.2 in TestFlight, and the problem still exists. You can't access the App if you connect to the Internet.

It might be due to a change lately on the payment provider I am using.

zhengzhengju, 2024-12-28:
@Clon1998 
Yes, it should be like this. So this will be fixed in v2.8.4, right?

Clon1998, 2024-12-28:
> @Clon1998 
> 
> Yes, it should be like this. So this will be fixed in v2.8.4, right?

I hopefully found the problem and fixed it in 2.8.4 but I can not test it and rely on you to give me feedback if it is working.

But you can not test it yet as we are waiting for 2.8.4 to get approved by apple.

zhengzhengju, 2024-12-28:
@Clon1998 
All right. There is another problem. In the application language, someone maliciously created non-standard naming, resulting in the inability to use zh-cn normally. Please refer to the screenshot below. It is recommended that you delete it or modify it. I tried to edit in place, but I don't know if I can pass!
[image]

Clon1998, 2024-12-28:
> @Clon1998
>
> All right. There is another problem. In the application language, someone maliciously created non-standard naming, resulting in the inability to use zh-cn normally. Please refer to the [...]

Mhh.
I am able to use all three mandarin language variations
- Chinese simplified
- Chinese traditional
- Chinese traditional, Honk Kong

Clon1998, 2025-01-02:
@zhengzhengju just received a Mail that the latest beta 2.8.4 was approved.
Can you verify for me if its working now?

Note, that the version might contain some styling issues tho.

zhengzhengju, 2025-01-03:
@Clon1998 
Hi, I have used the 2.8.4 beta version. The startup problem has been solved. Strangely enough, 2.8.3 can also be started now. I think the problem may not be in the program, but in the integrated in-app purchase provider function.

Clon1998, 2025-01-03:
> @Clon1998 
> 
> Hi, I have used the 2.8.4 beta version. The startup problem has been solved. Strangely enough, 2.8.3 can also be started now. I think the problem may not be in the program, but in the integrated in-app purchase provider function.

Thanks for the feedback.
I am glad it's working again.

zhengzhengju, 2025-01-03:
@Clon1998 
Ok, looking forward to the official version of 2.8.4. Looking forward to fixing the problem and the language problem. Thank you!

closed 2025-01-04 by Clon1998