Clon1998/mobileraker #449: Having a filament sensor in printer.cfg causes the miscellaneous card to disappear from the UI

RK392, 2024-12-21:
## Bug Report

### Description
As of a recent update, the miscellaneous card disappeared from the UI. After investigating, it turns out that removing the filament sensor from printer.cfg restores the miscellaneous card giving me access to my output pins.

### Steps to Reproduce
1. Add a filament sensor to your printer.cfg
2. Refresh the printer on Mobileraker
3. Observe the missing miscellaneous card
4. Remove the filament sensor from your printer.cfg
5. Refresh the printer on Mobileraker
6. Observe the miscellaneous card returning if you have output pins

### Expected Behavior
The miscellaneous card should be visible on the UI.

### Version Information

- Device-OS (Android/iOS): Android 13
- Mobileraker: 2.8.3-583
- Klipper: v0.12.0-404-g80d185c9
- Moonraker: v0.9.3-3-gccfe32f2

### Debug Logs
Logs are available upon request.

### Additional Context
The miscellaneous card was visible earlier, including the filament sensor box showing whether filament is being detected. Also, Mobileraker still shows a "Filament Sensor Triggered" dialog box even with the missing miscellaneous card which tells me that Mobileraker still recognizes the sensor.

Clon1998, 2024-12-21:
Hey,
I am unable to reproduce the issue. I have multiple printers with running filament sensors.

Can you please provide me the apps log file (App settings -> Scroll down -> logs) and the klipper config of the sensor

RK392, 2024-12-21:
Sure, here's the most recent log.

[mobileraker_2024-12-22T01-12-17.log](https://github.com/user-attachments/files/18220100/mobileraker_2024-12-22T01-12-17.log)

Thanks for your time and help!

Clon1998, 2024-12-22:
Hey @RK392 
Thanks for the log files. 
The problem was on my end due to a faulty code section related to sorting the available misc objects in the Misc. Card.

I will include the fix in the next release. For now, the only workarounds I can think of are not to include the filament sensor in the printer config (Not ideal...) or to include a `_` as a prefix to hide the object from the app

RK392, 2024-12-22:
Hey @Clon1998

Awesome! Thanks for the prompt response as usual!

commit 8100fa68 (2024-12-22): fix: Pins/Output card breaks if filamentsensor is used

Fix #449