Clon1998/mobileraker #452: Creality K2: custom WebRTC feed

jamincollins, 2024-12-26:
## Feature Request

### Problem Description
So, the Creality K2 has something of an odd/custom camera feed.
The camera is available on http://printer_ip:8000/

I can't seem to find a way in the current Mobileraker UI to reference this camera.

### Proposed Solution
Copy the logic of ether go2rtc or camerastreamer webrtc code and call to `/call/webrtc_local` and thats it

### Alternatives Considered

### Additional Context
Here is the script from the above referenced page that shows the call to `/call/webrtc_local`

```
     <script> 
       var pc = new RTCPeerConnection({ 
         iceServers: [{urls: 'stun:stun.l.google.com:19302'}] 
       }); 
       var log = msg => { console.log(msg); }; 
       function sendOfferToCall(sdp) { 
         var xhttp = new XMLHttpRequest(); 
         xhttp.onreadystatechange = function() { 
           if (this.readyState == 4 && this.status == 200) { 
             let res = JSON.parse(atob(this.responseText)); 
             console.log(res); 
             if(res.type == 'answer') { 
               pc.setRemoteDescription(new RTCSessionDescription(res)); 
             } 
           } 
         }; 
         xhttp.open('POST', '/call/webrtc_local'); 
         xhttp.setRequestHeader('Content-Type', 'plain/text'); 
         xhttp.send(btoa(JSON.stringify({'type': 'offer', 'sdp': sdp}))); 
       } 
       pc.ontrack = function (event) { 
         var el = document.getElementById('remoteVideos'); 
         el.srcObject = event.streams[0]; 
         el.autoplay = true; 
         el.controls = true; 
         el.muted = true; 
       }; 
       pc.oniceconnectionstatechange = e => log(pc.iceConnectionState); 
       pc.onicecandidate = event => { 
         if(event.candidate === null) sendOfferToCall(pc.localDescription.sdp) 
       }; 
       pc.addTransceiver('video', {'direction': 'sendrecv'}) 
       pc.createOffer().then(d => pc.setLocalDescription(d)).catch(log); 
     </script> ```

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [x] Provided a clear description of the problem you're facing.
- [X] Clearly outlined the proposed solution or feature.
- [ ] Mentioned any alternative solutions or ideas you've considered.
- [X] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Clon1998, 2024-12-26:
I already implemented multiple webRtc sources. Just open up the printer settings in the app and add a webcam

[image]

jamincollins, 2024-12-27:
I've tried them, none of them seem to work for the Creality K2.  Happy to provide whatever information will help.

jamincollins, 2024-12-27:
[image]

[image]

Clon1998, 2024-12-27:
Can you provide me the log files?
Seems like something during the WebRtc handshake is failing due to a type error.

Clon1998, 2024-12-27:
Hey @jamincollins 
I had another look at the code you posted. Is this part of either the Mainsail or Fluidd project or is it a code section of the modified Mainsail fork of Creality and they are using (once again) a custom solution to annoy everyone?

If later is the case, I will need to add a custom WebRtc Manager to support their handshake format.
Can you provide me a network trace from your browser so I can replicate their messaging commands without owning/having access to a K2?

jamincollins, 2024-12-27:
The camera is a completely stand alone app/binary on the K2, `/usr/bin/webrtc`.  The snippet is from the web page buried in this app/binary.

Here's the recording you requested, but it amounts to the same thing I detailed above:
[k2_Archive [24-12-27 12-08-10].har.zip](https://github.com/user-attachments/files/18262948/k2_Archive.24-12-27.12-08-10.har.zip)

Clon1998, 2024-12-27:
> The camera is a completely stand alone app/binary on the K2, `/usr/bin/webrtc`.  The snippet is from the web page buried in this app/binary.
>
>
>
> Here's the recording you requested, but it amounts to [...]

I will look into it. If it's really just these three requests it should be an easy thing to add.

jamincollins, 2024-12-27:
It should be, the only other possible niggle may be scaling the feed you get, but I suspect that's already handled in your existing code.

jamincollins, 2025-01-07:
Just wanted to check in on this.

Clon1998, 2025-01-07:
Hey @jamincollins 
Thanks for the reminder, I actually started to implement this right now. 
I can hopefully finish the implementation as it seems like creality is just using a `camera-streamer` and all requests/responses are Base64 encoded.

I will just add a new type that can handle this.

Clon1998, 2025-01-07:
I have triggered a new build on codemagic.
I will let you know once it is done and provide you an APK via GitHub

Clon1998, 2025-01-07:
Can you test the new creality apk on the release page:

https://github.com/Clon1998/mobileraker/releases/tag/android2.8.4

Clon1998, 2025-01-08:
@jamincollins can you give me feedback if the new webcam type is working?

I tried it with my standard klipper printer and it seems like the mentioned path is also available. However, my camera-streamer responds with a `405` code.

jgmdean, 2025-01-09:
Happy to test this on iPhone. 

Oddly enough Safari, Chrome, and Firefox on MacOS all show the stream when accessing the :8000 URL directly. Ditto Safari on iPhone.

Clon1998, 2025-01-09:
> Happy to test this on iPhone. 
> 
> 
> 
> Oddly enough Safari, Chrome, and Firefox on MacOS all show the stream when accessing the :8000 URL directly. Ditto Safari on iPhone. 

Eure, I just added the build to TestFlight which is now awaiting approval from apple. So it might take up to 24hrs

https://testflight.apple.com/join/h9JwuyR0

Clon1998, 2025-01-09:
> Sorry for the delay.
> 
> 
> 
> I just tested this.  I see the `WebRtcCreality`, and I've set the streaming URL to `http://k2:8000`.
> 
> 
> 
> I get:
> 
> ```
> 
> FormatException: Invalid character (at character 1)
> 
> <!DOCTYPE html>
> 
> ^
> 
> ```

Can you provide me the device logs

Clon1998, 2025-01-09:
Also try to include the "call/webrtc_local" in the webcam url

jamincollins, 2025-01-10:
[Most recent Mobileraker logs.txt](https://github.com/user-attachments/files/18368877/Most.recent.Mobileraker.logs.txt)

jgmdean, 2025-01-10:
> > Happy to test this on iPhone.
> 
> Eure, I just added the build to TestFlight which is now awaiting approval from apple. So it might take up to 24hrs
> 
> https://testflight.apple.com/join/h9JwuyR0

No luck. 
[image]
[image]

> > > Happy to test this on iPhone.
> > 
> > 
> > Eure, I just added the build to TestFlight which is now awaiting approval from apple. So it might take up to 24hrs
> > https://testflight.apple.com/join/h9JwuyR0
> > No Luck:

O have the modification from the script at https://github.com/jamincollins/k2-improvements
To get the webcam working in Fluidd. Test shouldn’t make a difference, right?

Clon1998, 2025-01-11:
@jamincollins 

You did not set the correct cam path.
Either use a relative address `/call/webrtc_local` or use an absolute address that includes the path like `http://k2:8000/call/webrtc_local` for the stream URL.

jamincollins, 2025-01-11:
Not sure how the relative is going to find the right port, but here they are:
[k2_8000.txt](https://github.com/user-attachments/files/18386335/k2_8000.txt)
[k2_relative.txt](https://github.com/user-attachments/files/18386336/k2_relative.txt)

jamincollins, 2025-01-11:
Here's the changes that were made to Fluidd if that perhaps helps:
[fluidd.patch.gz](https://github.com/user-attachments/files/18386408/fluidd.patch.gz)

Clon1998, 2025-01-11:
> Not sure how the relative is going to find the right port, but here they are: [k2_8000.txt](https://github.com/user-attachments/files/18386335/k2_8000.txt) [k2_relative.txt](https://github.com/user-attachments/files/18386336/k2_relative.txt)

Seems like you're cam config within mobileraker is still wrong as both versions still use the same URI:
`💡 [CrealityRtcManager#660642722(http://k2:8000/)] Sending offer to cam at http://k2:8000/`

[image]
[image]

jamincollins, 2025-01-11:
[image]

Clon1998, 2025-01-11:
Open sidebar -> Tap gear icon at the top -> webcam section -> click on webcam name -> change stream url to absolute path -> hit save

[image]

jamincollins, 2025-01-12:
[image]
[image]
[k2_relative_take2.txt](https://github.com/user-attachments/files/18391272/k2_relative_take2.txt)

jamincollins, 2025-01-12:
[image]
[image]
[k2_fully_scoped.txt](https://github.com/user-attachments/files/18391285/k2_fully_scoped.txt)

jamincollins, 2025-01-12:
[image]
[image]
[k2_just_port .txt](https://github.com/user-attachments/files/18391294/k2_just_port.txt)

jgmdean, 2025-01-13:
> @jamincollins
> 
> You did not set the correct cam path. Either use a relative address `/call/webrtc_local` or use an absolute address that includes the path like `http://k2:8000/call/webrtc_local` for the stream URL.
[image]

http://192.168.123.39:8000/ 
works in both fluid and in a browser directly

Clon1998, 2025-01-13:
> [image]
>
>
>
> http://192.168.123.39:8000/
>
> works in both fluid and in a browser directly

For nearly all webcam config setups I made them as customizable as possible. That's why for the K2 cam I once again did not include code to append the relative path to its call url.

The error is a different one again. Can you provide me the log file of the app so I can have a look where the type error is in my code.

jamincollins, 2025-01-13:
I've provided this version and log file

Clon1998, 2025-01-14:
> I've provided this version and log file [#452 (comment)](https://github.com/Clon1998/mobileraker/issues/452#issuecomment-2585925738)

Thanks,
i was able to locate the issue and fix it.
I will rebuild a new APK and IOS version for you guys to try.

Clon1998, 2025-01-14:
@jamincollins please give the [creality-webrtc-v2.apk](https://github.com/Clon1998/mobileraker/releases/download/android2.8.4/creality-webrtc-v2.apk) a try.

jamincollins, 2025-01-14:
[image]
[mobileraker_2025-01-14T11-10-07.log](https://github.com/user-attachments/files/18414475/mobileraker_2025-01-14T11-10-07.log)

Clon1998, 2025-01-14:
@jamincollins [creality-webrtc-v3.apk](https://github.com/Clon1998/mobileraker/releases/download/android2.8.4/creality-webrtc-v3.apk).

Ensured I actually use the SDP constraints I defined lol....

jamincollins, 2025-01-14:
Now it continually says "Trying to connect...":
[k2_trying_to_connect.txt](https://github.com/user-attachments/files/18415229/k2_trying_to_connect.txt)

Clon1998, 2025-01-14:
> Now it continually says "Trying to connect...": [k2_trying_to_connect.txt](https://github.com/user-attachments/files/18415229/k2_trying_to_connect.txt)

Okay, I will verify that tomorrow. Might be that the webrtc lib I am using simply does not support it.
The log states that the webrtc handling is done. So by default webrtc will do its magic and start working.

Clon1998, 2025-01-17:
Maybe I am blind but I can not find what else besides a platform/lib limitation is the error now:

```dart
/*
 * Copyright (c) 2023-2025. Patrick Schmidt.
 * All rights reserved.
 */

import 'dart:async';
import 'dart:convert';

import 'package:common/exceptions/mobileraker_exception.dart';
import 'package:common/util/logger.dart';
import 'package:dio/dio.dart';
import 'package:flutter_webrtc/flutter_webrtc.dart';

import 'data/webrtc_connection_event.dart';
import 'webrtc_manager.dart';

class CrealityRtcManager implements WebRtcManager {
  final Map<String, dynamic> offerSdpConstraints = {
    'mandatory': {
      'OfferToReceiveAudio': false,
      'OfferToReceiveVideo': true,
    },
    'optional': [],
  };

  final List<Map<String, String>> iceServers = [
    {'url': 'stun:stun.l.google.com:19302'}
  ];

  CrealityRtcManager({required Dio dio, required Uri camUri})
      : _camUri = camUri,
        _dio = dio;

  final Dio _dio;

  final Uri _camUri;

  final RTCVideoRenderer _localRenderer = RTCVideoRenderer();
  String? _remoteId;
  RTCPeerConnection? _pc;

  final StreamController<WebRtcConnectionEvent> _streamController = StreamController();

  @override
  Stream<WebRtcConnectionEvent> get stream => _streamController.stream;

  RTCPeerConnectionState? __connectionState;

  set _connectionState(RTCPeerConnectionState state) {
    if (state == __connectionState) return;
    logger.i(
        '[CrealityRtcManager#${identityHashCode(this)}($_camUri)] RTC _onConnectionState: $__connectionState -> $state');
    __connectionState = state;
    WebRtcConnectionEvent event = switch (state) {
      RTCPeerConnectionState.RTCPeerConnectionStateClosed => WebRtcConnectionEventClosed(),
      RTCPeerConnectionState.RTCPeerConnectionStateFailed => WebRtcConnectionEventFailed(),
      RTCPeerConnectionState.RTCPeerConnectionStateDisconnected => WebRtcConnectionEventDisconnected(),
      RTCPeerConnectionState.RTCPeerConnectionStateNew => WebRtcConnectionEventNew(),
      RTCPeerConnectionState.RTCPeerConnectionStateConnecting => WebRtcConnectionEventConnecting(),
      RTCPeerConnectionState.RTCPeerConnectionStateConnected => WebRtcConnectionEventConnected(_localRenderer),
    };
    if (!_streamController.isClosed) _streamController.add(event);
  }

  @override
  Future<void> startCam() async {
    logger.i('[CrealityRtcManager#${identityHashCode(this)}($_camUri)] Starting Camera connection');
[... 179 lines cut ...]
          e);
      // Some versions of the cam server do not support ICE candidates
    }
  }

  void _onIceConnectionState(RTCIceConnectionState event) {
    logger.i('[CrealityRtcManager#${identityHashCode(this)}($_camUri)] RTC onIceConnectionState: $event');
  }

  void _onSignalingState(RTCSignalingState state) {
    logger.i('[CrealityRtcManager#${identityHashCode(this)}($_camUri)] RTC onSignalingState: $state');
  }

  void _onConnectionState(RTCPeerConnectionState event) {
    _connectionState = event;
  }

  @override
  void dispose() {
    stopCam().ignore();
    _localRenderer.dispose().ignore();
    _streamController.close().ignore();
  }
}
```

jgmdean, 2025-01-19:
Is there a new iOS build you want me to test via TestFlight? 

I now have two K2’s so I can test two configs at once. 😎

How do I provide the log file in iOS?

[image]

Clon1998, 2025-02-02:
Hey,
I had another look at the log you provided @jamincollins and just triggered a new build with some minor changes including more logs and a new webrtc lib version.

I will upload it once the build is done. Can you please let me know if it changes anything?

Clon1998, 2025-02-02:
Please give this [version](https://github.com/Clon1998/mobileraker/releases/download/android2.8.4/creality-webrtc-v4.apk) a try.

If it does not work, the app logs and the webrtc (camera-streamer) logs would also be awesome

jamincollins, 2025-02-02:
It now appears to just hang at "Trying to connect...", here are the logs:

[mobileraker_2025-02-02T14-17-13.log](https://github.com/user-attachments/files/18634018/mobileraker_2025-02-02T14-17-13.log)
[mobileraker_2025-02-02T14-20-55.log](https://github.com/user-attachments/files/18634017/mobileraker_2025-02-02T14-20-55.log)

[cam_app.log](https://github.com/user-attachments/files/18634024/cam_app.log)
[cam_sub_app.log](https://github.com/user-attachments/files/18634023/cam_sub_app.log)

jamincollins, 2025-02-08:
Does this help at all, this is the actual working implementation from the patched Fluidd:

[WebrtcCrealityk2RtcCamera.vue.zip](https://github.com/user-attachments/files/18716192/WebrtcCrealityk2RtcCamera.vue.zip)

Clon1998, 2025-02-08:
> Does this help at all, this is the actual working implementation from the patched Fluidd:
> 
> [WebrtcCrealityk2RtcCamera.vue.zip](https://github.com/user-attachments/files/18716192/WebrtcCrealityk2RtcCamera.vue.zip)

I have compared it to mine which I posted above and sadly I can not spot anything that I am missing here in terms of handling web etc signaling.

SCiunczyk, 2025-02-13:
Hi, if I can help somehow, let me know. I have same symptoms as @jamincollins

Clon1998, 2025-02-16:
> Hi, if I can help somehow, let me know. I have same symptoms as @jamincollins 

I've tried everything I could besides buying a K2 myself to have physical access to the webcam feed to debug it myself.

I also posted the code I am using above. I am not an expert myself on webrtc but based on my understanding I am handling the messaging correctly and just one small thing is missing to get it working.
I don't know if the logs on the K2 mention the thing that I am missing or not but I am seriously out of ideas.

closed 2025-02-17 by Clon1998

nagyrobi, 2025-04-08:
@Clon1998 apparently K2 doesn't send a standard SDP: https://github.com/AlexxIT/go2rtc/issues/1600#issuecomment-2783606504

briankelly, 2025-04-10:
Just dropping in on this issue since I saw it was reopened. I'm running an Ender-3 V3 KE with the Creality Webcam (CCX2F3298) and Nebula pad. I believe these are the same devices on the K2.

I've rooted my KE and I've gotten pretty familiar with the various services running on it. Happy to help debug this issue as I would also like to migrate from using mjpegstreamer to using webrtc (or any kind of h.264 streaming) with mobileraker. (Thanks for the app by the way -- happy iOS user here.)

thx1200, 2025-04-12:
Hello, I just got a Creality Hi, which uses the exact same custom camera nonsense as the K2.  I am a developer and can assist in troubleshooting and possibly some coding if necessary, although my knowledge of WebRTC and video codecs is very limited.

Clon1998, 2025-04-12:
I'll post the code again or crate a simple demo app that contains it.
This should make it easier for you guys to debug the webcam code.

nagyrobi, 2025-04-12:
[Apparently](https://github.com/Guilouz/Creality-Helper-Script-K2/issues/3#issuecomment-2797868763) K1 is getting the same firmware series as K2...

Clon1998, 2025-04-12:
> [Apparently](https://github.com/Guilouz/Creality-Helper-Script-K2/issues/3#issuecomment-2797868763) K1 is getting the same firmware series as K2...

Good to know

Clon1998, 2025-04-16:
If anyone wants to give that thing a try. This should be a minimum example:
https://github.com/Clon1998/creality-flutter-webrtc

cc @thx1200

nagyrobi, 2025-04-16:
Have also an apk for us to sideload?

thx1200, 2025-04-20:
I did some more testing and I'm now not entirely sure the HI uses the same camera or codec.  Although it has its non-standard thing on port 8000, it only works on pc.  I can't get it to stream directly on my phone.  I also installed go2rtc which doesn't recognize the codec either.

If a generalized Creality solution for this comes about, I'm still happy to test but I'm not sure my contributions with my Hi are helpful now for the K2 issue specifically, though I'm sure they are closely related.  I ended up buying a cheap $10 webcam that has a better angle on my prints anyway.  I'm having a different issue (more minor) with that but I'll open a separate issue so as not to big down this thread. 

If you need testing on the HI, I'm happy to help in the future.

thx1200, 2025-04-21:
I just had a thought on this that might be the easiest way to solve it.  Since what Creality gives us at port 8000 is a web page that loads up a js video player, can you add a camera mode that just is basically an embedded WebView that points to that URL?  The http://printer:8000/ will just end up being a full screen webcam and it renders correctly in WebView on mobile, so that would basically make this a super easy way to embed webcams without having to decode the stream.

thx1200, 2025-04-22:
Alright, spent a few hours learning how to set up Android Studio and Flutter (wow it's ... a lot) and got the app to compile.  I have some results to share.  It doesn't work on my Hi.  But I can see the "fixup" correctly removing the codec.  It just never stops spinning and the debug stream just shows 0 frames coming in.

Debug lines are all like this:
`I/org.webrtc.Logging( 9483): EglRenderer: Duration: 4005 ms. Frames received: 0. Dropped: 0. Rendered: 0. Render fps: .0. Average render time: NA. Average swapBuffer time: NA.`

The SDP stuff starts out like this:
`v=0
o=- 1495799811084970 1495799811084970 IN IP4 0.0.0.0
s=-
t=0 0
a=msid-semantic:WMS *
a=group:BUNDLE 0
m=video 9 UDP/TLS/RTP/SAVPF 96 104
a=rtcp-fb:104 nack
a=rtcp-fb:104 nack pli
a=fmtp:96 profile-level-id=42e01f;level-asymmetry-allowed=1
a=fmtp:104 profile-level-id=42e01f;packetization-mode=1;level-asymmetry-allowed=1
a=fmtp:104 x-google-max-bitrate=6000;x-google-min-bitrate=2000;x-google-start-bitrate=4000
a=rtpmap:96 H264/90000
a=rtpmap:104 H264/90000
a=ssrc:1 cname:pear
c=IN IP4 0.0.0.0
a=sendonly
a=mid:0
a=rtcp-mux
a=ice-ufrag:f5HR
a=ice-pwd:7dX7C+G4qja620MobhaTgp
a=ice-options:trickle
a=fingerprint:sha-256 67:34:8E:65:0D:3D:C6:CC:6F:D3:0B:F3:39:3E:C1:FB:F1:A3:D9:B5:9B:17:97:24:57:31:4E:67:02:9C:F7:AF
a=setup:passive
a=candidate:1 1 UDP 2015363327 192.168.10.233 53209 typ host
a=candidate:2 1 TCP 1015021823 192.168.10.233 0 typ host tcptype active
a=candidate:3 1 TCP 1010827519 192.168.10.233 51495 typ host tcptype passive`

And ends up like this after the fixup:
`v=0
o=- 1495799811084970 1495799811084970 IN IP4 0.0.0.0
s=-
t=0 0
a=msid-semantic:WMS *
a=group:BUNDLE 0
m=video 9 UDP/TLS/RTP/SAVPF 104
a=rtcp-fb:104 nack
a=rtcp-fb:104 nack pli
a=fmtp:104 profile-level-id=42e01f;packetization-mode=1;level-asymmetry-allowed=1
a=rtpmap:104 H264/90000
a=ssrc:1 cname:pear
c=IN IP4 0.0.0.0
a=sendonly
a=mid:0
a=rtcp-mux
a=ice-ufrag:f5HR
a=ice-pwd:7dX7C+G4qja620MobhaTgp
a=ice-options:trickle
a=fingerprint:sha-256 67:34:8E:65:0D:3D:C6:CC:6F:D3:0B:F3:39:3E:C1:FB:F1:A3:D9:B5:9B:17:97:24:57:31:4E:67:02:9C:F7:AF
a=setup:passive
a=candidate:1 1 UDP 2015363327 192.168.10.233 53209 typ host
a=candidate:2 1 TCP 1015021823 192.168.10.233 0 typ host tcptype active
a=candidate:3 1 TCP 1010827519 192.168.10.233 51495 typ host tcptype passive`

And the app just spins...

[image]

Going to keep digging into it to see if I can figure anything out but this is way outside my area of expertise lol.

Clon1998, 2025-07-24:
> Hi. Any news on this? Few days ago K2 received new firmware but that didnt change anything, still cant get webcam.

Nope.
I might need get my hands on a new creality Machine that uses a similar firmware to have a device to reproduce the issue locally.

duka10, 2025-07-25:
> > Ok great. Im willing to test it when you release it.
> 
> Do you happen to know which creality printers share the firmware with the K2.
> 
> Does the K1 also use the K2 klipper fork?

Yes i read that someone mention that. I have both of them and i dont think they share same firmware.

commit cc62a0a7 (2025-01-07): feat: Add Creality WebRtc

feat: Add Creality WebRtc

Fix #452