Clon1998/mobileraker #458: Cannot set to Chinese (Simplified)

pfdr2333, 2025-01-01:
The actual effective language of the Chinese/Chinese (Simplified)/Chinese (Traditional) options is Chinese (Traditional)

The mobileraker version is 2.8.3

System version: HyperOS 1.0.10.0 for Android 14

System language: Chinese (Simplified)

[image]

Clon1998, 2025-01-01:
Hey,
I should have addressed that in the latest beta version.

I can hopefully publish this fix alongside other hotfixes next week.

zhengzhengju, 2025-01-01:
@Clon1998 
Hi, I'm glad I'm done with all the simplified Chinese translations. Looking forward to your approval and application to the next version update! I also look forward to your addition to the list of contributors! Thank you

closed 2025-01-04 by Clon1998