Clon1998/mobileraker #464: Artillery Sidewinder X4 Plus S1 - Connection Freezes / Crashes

vaaarg, 2025-01-18:
Hi. Just wanted to try the Mobileraker App, but got several errors while trying to connect. This is the second attempt to use it. First Time printer freezed completely while printing. Just wanted to leave a note or wanted to ask if s.o. could help to transalte the error. 

Printer : Artillery Sidewinder X4 Plus S1

`Exception:
 JRpcTimeoutError: JRpcMethod(printer.objects.query) timed out after 5 seconds

#0      JsonRpcClient.sendJRpcMethod.<anonymous closure> (package:common/network/json_rpc_client.dart:197)
#1      FutureExtensions.onError.onError (dart:async/future.dart:1068)
#2      _RootZone.runBinary (dart:async/zone.dart:1681)
#3      _FutureListener.handleError (dart:async/future_impl.dart:219)
#4      Future._propagateToListeners.handleError (dart:async/future_impl.dart:914)
#5      Future._propagateToListeners (dart:async/future_impl.dart:935)
#6      Future._completeError (dart:async/future_impl.dart:715)
#7      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:987)
#8      TickerFuture.whenCompleteOrCancel.thunk (package:flutter/src/scheduler/ticker.dart:452)
#9      _Timer._runTimers (dart:isolate-patch/timer_impl.dart:398)
#10     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:429)
#11     _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:184)`

After this errror occures also the web based fluidd connection was lost until i rebooted the printer.

Another log:

`10:13:46.186 (+0:00:00.000092)
💡 -----------------------
10:13:46.186 (+0:00:00.000266)
💡 Starting Mobileraker...
10:13:46.186 (+0:00:00.000308)
💡 -----------------------
10:13:46.186 (+0:00:00.000882)
💡 AppLifecycleState changed from null to AppLifecycleState.resumed
10:13:46.187 (+0:00:00.001712)
💡 *****************************
10:13:46.187 (+0:00:00.001826)
💡 Mobileraker is warming up...
10:13:46.201 (+0:00:00.015424)
💡 Mobileraker Version: PackageInfo(appName: Mobileraker, buildNumber: 588, packageName: com.mobileraker.android, version: 2.8.4, buildSignature: CFEF03C44FB49BBFE6587CF518DD1C62B88F564D3BF0109BBB77C0C8B21EF41F, installerStore: com.android.vending)
10:13:46.201 (+0:00:00.015565)
💡 *****************************
10:13:46.202 (+0:00:00.016038)
💡 Warmup provider changed from StartUpStep.firebaseCore to StartUpStep.firebaseAppCheck
10:13:46.232 (+0:00:00.046542)
💡 Warmup provider changed from StartUpStep.firebaseAppCheck to StartUpStep.firebaseRemoteConfig
10:13:46.276 (+0:00:00.090620)
💡 Completed FirebaseRemote init
10:13:46.276 (+0:00:00.090930)
💡 Warmup provider changed from StartUpStep.firebaseRemoteConfig to StartUpStep.firebaseAnalytics
10:13:46.277 (+0:00:00.091299)
💡 Warmup provider changed from StartUpStep.firebaseAnalytics to StartUpStep.firebaseAuthUi
10:13:46.279 (+0:00:00.093864)
💡 Firebase Auth User changed from null to AsyncLoading<User?>()
10:13:46.280 (+0:00:00.094011)
💡 Warmup provider changed from StartUpStep.firebaseAuthUi to StartUpStep.hiveBoxes
10:13:46.280 (+0:00:00.094141)
💡 Firebase Auth User changed from AsyncLoading<User?>() to AsyncData<User?>(value: null)
10:13:46.286 (+0:00:00.100188)
💡 FirebaseRemote values are fetched and activated! The last fetch was not successful and on 2025-01-18 09:19:42.893
10:13:46.296 (+0:00:00.110261)
💡 Completed Hive init
10:13:46.296 (+0:00:00.110364)
💡 Warmup provider changed from StartUpStep.hiveBoxes to StartUpStep.easyLocalization
10:13:46.297 (+0:00:00.111237)
💡 Warmup provider changed from StartUpStep.easyLocalization to StartUpStep.paymentService
10:13:46.319 (+0:00:00.133335)
💡 Fetching customer info
10:13:46.321 (+0:00:00.135478)
💡 Selecting machine null
10:13:46.322 (+0:00:00.136291)
💡 Setting device version to Purchases: PackageInfo(appName: Mobileraker, buildNumber: 588, packageName: com.mobileraker.android, version: 2.8.4, buildSignature: CFEF03C44FB49BBFE6587CF518DD1C62B88F564D3BF0109BBB77C0C8B21EF41F, installerStore: com.android.vending)
[... 567 lines cut ...]
10:14:40.378 (+0:00:54.192760)
💡 DashboardPageController: (aIdx: 0, l:  2) -> (aIdx: 1, l:  2)
10:14:40.380 (+0:00:54.194548)
💡 Rebuilding _PrinterAppBar for Mein Drucker
10:14:41.489 (+0:00:55.303103)
💡 [BottomNavigationBar] Page Changed: 1
MobilerakerDioException [DioExceptionType.connectionError]: The connection errored: No route to host This indicates an error which most likely cannot be solved by the library.
Error: SocketException: No route to host (OS Error: No route to host, errno = 113), address = 192.168.178.99, port = 40796
10:14:44.974 (+0:00:58.788029)
⚠️ DioException while requesting MJPEG-Stream
10:14:47.595 (+0:01:01.409510)
💡 Navigating to /setting
10:14:47.604 (+0:01:01.418201)
💡 [MobilerakerRouteObserver-Main]::PUSH Instance of '_PageBasedMaterialPageRoute<void>'.
10:14:48.435 (+0:01:02.249632)
⚠️ [Mein Drucker (d5b5c733-779c-4691-83f7-f4a10df55423)@http://192.168.178.99] Unable to propagate new notification settings because JRPC was not connected!
MobilerakerException{Machine not connected, parentException: null, parentStack: null}
10:14:48.435 (+0:01:02.249936)
⚠️ Error while trying to fetch CompanionMeta for machine Mein Drucker (d5b5c733-779c-4691-83f7-f4a10df55423)
10:14:49.850 (+0:01:03.664711)
💡 Show Dialog request for DialogType.logging
10:14:49.851 (+0:01:03.665619)
💡 [MobilerakerRouteObserver-Main]::PUSH Instance of 'DialogRoute<DialogResponse<dynamic>>'.
10:15:00.798 (+0:01:14.612655)
💡 StreamMjpegManager DISPOSED`

Clon1998, 2025-01-18:
Hey,

that's an timeout error that occurs if your moonraker server, which is the web services running on your klipper host, does not answer in your specified timeout.

You can increase it to whatever value you want, 30 for example. Just open the sidebar -> tap the gear icon at the top -> Update the  **Client Timeout** to 15 s -> hit save.

Regarding your entire printer freezing. What SBC are you using to run klipper/moonraker/fluidd? An Pi 3?
In some cases these have very limited resources and having fluidd + mobileraker open can cause issues.

vaaarg, 2025-01-18:
> Regarding your entire printer freezing. What SBC are you using to run klipper/moonraker/fluidd? An Pi 3? In some cases these have very limited resources and having fluidd + mobileraker open can cause issues.

[Sidewinder X4 Plus S1](https://www.artillery3d.com/de-eu/products/sidewinder-x4-plus-s1)

As far as i know everything is embedded within the X4 Plus S1, i'm just opening the device ip and fluidd surface starts

Clon1998, 2025-01-18:
> > Regarding your entire printer freezing. What SBC are you using to run klipper/moonraker/fluidd? An Pi 3? In some cases these have very limited resources and having fluidd + mobileraker open can cause issues.
> 
> [Sidewinder X4 Plus S1](https://www.artillery3d.com/de-eu/products/sidewinder-x4-plus-s1)
> 
> As far as i know everything is embedded within the X4 Plus S1, i'm just opening the device ip and fluidd surface starts.
> 
> I tried to use the api key provided within fluidd, that didnt change sth. Will try the timeout hint. thx
> 
> Edit. Timeout didnt change it. If i start Mobileraker "Solo" the Printer Wifi Connection freezes after few seconds and printer is not reachable until reboot.

Okay,
I will have another look at the log file. But to me it seems like the printer's SBC is underpowered or uses a very old custom moonraker version that does not like some of the newer calls. 
Can you do me a favor and also upload the moonraker log files of an occasion where the printer needed a restart?

vaaarg, 2025-01-18:
[https://drive.google.com/drive/folders/14gyHJVu8VT-c9a1i5AuNA9Ownvvl6ei0](klipper and moonraker Logs)

Thank You for your effort

Clon1998, 2025-01-19:
> [Logs](https://drive.google.com/drive/folders/14gyHJVu8VT-c9a1i5AuNA9Ownvvl6ei0)
> 
> Thank You for your effort

To me it seems like the SBC has not enough power.

Also the moonraker version is really old. It is running software_version: v0.7.1 vs v0.9.3 which is the latest.

closed 2025-03-08 by Clon1998