Clon1998/mobileraker #465: Error while loading the language files, smartphone language in catalan.

Gerqp10, 2025-01-19:
## Bug Report

### Description
I cannot use the app if the smartphone main language is in Catalan.

### Steps to Reproduce
1. Open the app
2. See the error.

### Expected Behavior
See the app interface and use it and have the main smartphone language in Catalan

### Screenshots

[image]
[image]
[image]

### Version Information

- Device-OS iOS): 18.2.1
- Mobileraker: 2.8.4
- Klipper: -
- Moonraker: -

### Debug Logs
I can't do it when the error comes up because I can't use the app. I share the logs of when I change the main language of the phone to English in case it is useful.

[mobileraker_2025-01-18T18-35-25.log](https://github.com/user-attachments/files/18470001/mobileraker_2025-01-18T18-35-25.log)
[mobileraker_2025-01-19T19-50-26.log](https://github.com/user-attachments/files/18470003/mobileraker_2025-01-19T19-50-26.log)
[mobileraker_2025-01-19T19-50-45.log](https://github.com/user-attachments/files/18470004/mobileraker_2025-01-19T19-50-45.log)
[mobileraker_2025-01-19T19-35-38.log](https://github.com/user-attachments/files/18470002/mobileraker_2025-01-19T19-35-38.log)
[mobileraker_2025-01-19T19-50-11.log](https://github.com/user-attachments/files/18470005/mobileraker_2025-01-19T19-50-11.log)

### Additional Context
My normal idiom preference order is 1-Catalan 2-Spanish 3-English.
I change the order to 1-English 2-Spanish 3-Catalan, an the app works fine, but i have to have the smartphone main language in english.
When entering the app in English, in the language menu i noticed that the Catalan language is there but with the Canadian flag.
I don't mind having the app in English, in fact I prefer it. but I would like to have the smartphone language in my native language.

Gerqp10, 2025-01-21:
> Hey, seems like a small error on my end due to the missing Catalan file. I will include a fix in the next release/ maybe even push a hotfix.

Thanks very much!!

oaragues, 2025-02-07:
> > Hey, seems like a small error on my end due to the missing Catalan file. I will include a fix in the next release/ maybe even push a hotfix.
> 
> Thanks very much!!

Hi. I'm in the exact same spot. Did you have any luck or still waiting for the hotfix? Or using your mobile in English rather than Català?
I don't know much of programming but in a vast majority of apps that doesn't have catalan, it switches to default which normally is english...

BTW I'm on android

Clon1998, 2025-02-07:
> Hi. I'm in the exact same spot. Did you have any luck or still waiting for the hotfix? Or using your mobile in English rather than Català?
> I don't know much of programming but in a vast majority of [...]

Hey,
sorry about that. Life got in the way as I am not developing Mobileraker full time.

The problem was that s1 requested Catala as language and after I approved it, did not provide any translation. This, caused an issue with the app as it already expected an Catalan language file which without any translation was not existing.

I removed Catalan for now. However, if anyone of you wants to help translate it. Please check the translation [guide](https://github.com/Clon1998/mobileraker/blob/master/docs/contribute_i18n.md).

I am still finalizing the last bits of the current release. If it takes longer I'll do an hotfix on Monday which only includes that Catalan fix.

- Patrick

oaragues, 2025-02-09:
Thank you very much!!

oaragues, 2025-02-09:
> Hey,
> sorry about that. Life got in the way as I am not developing Mobileraker full time.
>
> The problem was that s1 requested Catala as language and after I approved it, did not provide any [...]

Thank you very much

commit 765b5536 (2025-01-19): fix: Catalan translation missing.

Fixes #465
closed 2025-02-17 by Clon1998