Clon1998/mobileraker #466: Email is not verified

wikolii, 2025-01-25:
## Bug Report

### Description
I changed phones. On the new de device I tried restoring my purchase and was asked to create an account. I received the verification email and clicked the link, everything seemed to work but in the app it still says "Email is not verified" and restoring purchase shows an error 

### Steps to Reproduce
1. Go to 'Support the Dev!'
2. Click on 'Restore / Sign in'
3. Click on 'Restore Purchases'
4. See the error.

### Expected Behavior
Get supporter benefits

### Screenshots
[image]
### Version Information

- Device-OS (Android/iOS): [iOS]
- Mobileraker: [2.8.5-593]
- Klipper: [version]
- Moonraker: [version]

### Debug Logs

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

wikolii, 2025-01-25:
I'm on my phone and can't find the way to upload the logs using the app

Clon1998, 2025-01-25:
Can you please provide me with the log files? You can find them via: Settings Page -> Scroll to bottom -> Debug logs.

The email verification is not as critical, and it might be resolved by just restarting the app :).

wikolii, 2025-01-25:
Attached log files, thanks
[mobileraker_2025-01-25T07-37-05.log](https://github.com/user-attachments/files/18547528/mobileraker_2025-01-25T07-37-05.log)
[mobileraker_2025-01-24T10-10-41.log](https://github.com/user-attachments/files/18547529/mobileraker_2025-01-24T10-10-41.log)
[mobileraker_2025-01-24T13-54-31.log](https://github.com/user-attachments/files/18547525/mobileraker_2025-01-24T13-54-31.log)
[mobileraker_2025-01-24T16-57-05.log](https://github.com/user-attachments/files/18547527/mobileraker_2025-01-24T16-57-05.log)
[mobileraker_2025-01-25T07-20-01.log](https://github.com/user-attachments/files/18547526/mobileraker_2025-01-25T07-20-01.log)

wikolii, 2025-01-26:
Ok, i was using the Testflight version of the app. I downloaded the one front eh app store and was able to restore my supporter status. 
The "Email is not verified" message is till there, but is not a problem. Thanks!

closed 2025-01-26 by wikolii