Clon1998/mobileraker #467: Error using the camera stream (only at 1 of 2 devices)

tefracky, 2025-02-01:
## Bug Report

### Description
I have two Android devices (Google Pixel 9 with Android 16 Beta and Samsung Galaxy S10+ with Android 12). For both devices, the App works fine besides the webcam. Using the Google Pixel, both webcams work well, but using the Samsung Galaxy, one stream is not loading. Sometimes, only a snapshot is shown but with different FPS values. Additionally, the App crashes from time to time.
Attached two pictures from the "problematic" phone.
The left camera ("Links") normally works and only the right camera ("Rechts") makes problems. The camera settings of both devices are identical (see screenshots 3 and 4).

### Steps to Reproduce
1. Install the App
2. Configure a printer
3. Try to use the webcams

### Expected Behavior
The stream should load and show a real stream not only a snapshot.

### Screenshots
If applicable, add screenshots to help explain the problem.
1 and 2: Not loading camera or a "snapshot" with good fps.
3 and 4: Camera settings

[image]
[image]
[image]
[image]

### Version Information

- Device-OS (Android/iOS): Android
- Mobileraker: 2.8.4
- Klipper: v0.12.0-418-g0114d72a
- Moonraker: v0.9.3-20-g52781fbd
- Mobileraker on Server: v0.4.0-91-g1d1e2ebe

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

[mobileraker_2025-01-31T22-34-15.log](https://github.com/user-attachments/files/18629233/mobileraker_2025-01-31T22-34-15.log)
[mobileraker_2025-01-31T22-34-49.log](https://github.com/user-attachments/files/18629235/mobileraker_2025-01-31T22-34-49.log)
[mobileraker_2025-01-31T19-24-53.log](https://github.com/user-attachments/files/18629237/mobileraker_2025-01-31T19-24-53.log)
[mobileraker_2025-01-31T20-59-20.log](https://github.com/user-attachments/files/18629236/mobileraker_2025-01-31T20-59-20.log)
[mobileraker_2025-01-31T22-15-01.log](https://github.com/user-attachments/files/18629234/mobileraker_2025-01-31T22-15-01.log)

I think this is the most relevant part:
```
22:34:32.281 (+0:00:17.061170)
💡 Rebuilding Async Guard: WebcamCard-e1624417-f7d6-4832-8d29-a577666f7504 with AsyncError<bool>(error: Bad state: The provider allWebcamInfosProvider:AllWebcamInfosProvider#1fbce(null) was disposed during loading state, yet no value could be emitted., stackTrace: #0      FutureHandlerProviderElementMixin.dispose (package:riverpod/src/async_notifier/base.dart:467)
💡 #1      StreamProviderElement.dispose (package:riverpod/src/stream_provider/base.dart:180)
💡 #2      ProviderContainer._disposeProvider (package:riverpod/src/framework/container.dart:319)
💡 #3      ProviderScheduler._performDispose (package:riverpod/src/framework/scheduler.dart:137)
💡 #4      ProviderScheduler._task (package:riverpod/src/framework/scheduler.dart:89)
💡 #5      ProviderScheduler.vsync.<anonymous closure>.invoke (package:riverpod/src/framework/scheduler.dart:41)
💡 #6      _UncontrolledProviderScopeElement.build (package:flutter_riverpod/src/framework.dart:396)
💡 #7      ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:5631)
💡 #8      Element.rebuild (package:flutter/src/widgets/framework.dart:5347)
💡 #9      BuildScope._tryRebuild (package:flutter/src/widgets/framework.dart:2694)
💡 #10     BuildScope._flushDirtyElements (package:flutter/src/widgets/framework.dart:2753)
💡 #11     BuildOwner.buildScope (package:flutter/src/widgets/framework.dart:3048)
💡 #12     WidgetsBinding.drawFrame (package:flutter/src/widgets/binding.dart:1176)
💡 #13     RendererBinding._handlePersistentFrameCallback (package:flutter/src/rendering/binding.dart:475)
💡 #14     SchedulerBinding._invokeFrameCallback (package:flutter/src/scheduler/binding.dart:1397)
💡 #15     SchedulerBinding.handleDrawFrame (package:flutter/src/scheduler/binding.dart:1318)
💡 #16     SchedulerBinding._handleDrawFrame (package:flutter/src/scheduler/binding.dart:1176)
💡 #17     _invoke (dart:ui/hooks.dart:312)
💡 #18     PlatformDispatcher._drawFrame (dart:ui/platform_dispatcher.dart:427)
💡 #19     _drawFrame (dart:ui/hooks.dart:283)
💡 )
```

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [X] Provided a clear bug description.
- [X] Listed detailed steps to reproduce the issue.
- [X] Described the expected behavior.
- [X] Included the Mobileraker version you are using.
- [X] Attached Mobileraker's debug log files.
- [X] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2025-02-01:
Hey,
just to clarify, were the screenshots taken on the Samsung G10+? The first two both show the Rechte Cam, which you mentioned does not work.

tefracky, 2025-02-01:
Sorry for the confusion. All screenshots and logs are from the S10. I just mentioned the Pixel to show that all is working fine on another device.

If I open the app the camera shows always the same picture (like a snapshot), but it is not a stream. Even if a good fps is shown, the picture does not change.

Clon1998, 2025-02-02:
> Sorry for the confusion. All screenshots and logs are from the S10. I just mentioned the Pixel to show that all is working fine on another device.
>
> If I open the app the camera shows always the same [...]

Alright, seems like this is caused by the error entry you already identified. I will see if I am able to reproduce it with my test Galaxy S10 :)

Clon1998, 2025-02-02:
Can you do me a favor and provide a screen recording while starting the app on the S10?

I think I identified the issue but at the same time I am seriously confused why some of the dashboard components are loading before the so-called `PrinterProviderGuard`, which is a small code section that ensures that the printer and klippy data are available before even rendering anything on the dashboard, is ready...

Clon1998, 2025-02-02:
Please give this [version](https://github.com/Clon1998/mobileraker/releases/download/android2.8.4/creality-webrtc-v4.apk) a try.

If it does not work, the app logs and the webrtc (camera-streamer) logs would also be awesome

tefracky, 2025-02-02:
> Please give this [version](https://github.com/Clon1998/mobileraker/releases/download/android2.8.4/creality-webrtc-v4.apk) a try.

This version works well! Thank you very much. Do you need the logs anyway?

Clon1998, 2025-02-02:
> > Please give this [version](https://github.com/Clon1998/mobileraker/releases/download/android2.8.4/creality-webrtc-v4.apk) a try.
> 
> This version works well! Thank you very much. Do you need the logs anyway?

If it works for you, it's fine by me.

In case it won't work for the others in this thread, please provide me the error logs.

commit 6e1ace4a (2025-02-02): fix: Providers of webcamservice dispose to early

fix #467
closed 2025-02-17 by Clon1998