Clon1998/mobileraker #468: more themes or the possibility to customize with own images

exerkaoss, 2025-02-01:
I've noticed that ZeroG or BTT themes are missing.is it possible to have them or have a customized option to upload our own images?

Thank you so much

Clon1998, 2025-02-01:
Hey,
adding a new theme for a community is not a problem.

Can you provide me some screenshots or design references so I can add a ZeroG theme?

exerkaoss, 2025-02-01:
[image]
[image]
[image]
[image]
[image]

This are the ones for ZeroG and i will try and get the ones for BigTreeTech too

exerkaoss, 2025-02-01:
[image]

Regarding bigtreetech, the svg logo is the only thing ican find from mainsail themes OG github repo.

Thanks a lot for supporting my request! I love the app and have it installed everywhere :D

Clon1998, 2025-02-01:
> [image]
> [image]
> [image]
> [image]
> [image]
> 
> This are the ones for ZeroG and i will try and get the ones for BigTreeTech too

Are you part of the official ZeroG team or can get s1 involved regarding the permission to use the logos in the app?

exerkaoss, 2025-02-01:
> > [image]
> > [image]
> > [image]
> > [image]
> > [image]
> > This are the ones for ZeroG and i will try and get the ones for BigTreeTech too
> 
> Are you part of the official ZeroG team or can get s1 involved regarding the permission to use the logos in the app?

no, i'm not on either zerog or BTT, but i know ZeroG is an open source project so i think they wouldn't have a problem to give permission.
BTT already gave it to mainsail in order to have the theme up (i'm using it), so it shouldn't be a problem to get that permission as well.
Would you like me to try and reach to  them?

exerkaoss, 2025-02-01:
> Are you part of the official ZeroG team or can get s1 involved regarding the permission to use the logos in the app?

[image]
[image]
Update, we have the go from ZeroG