Clon1998/mobileraker #469: Exclude objects

DBa20-24, 2025-02-03:
## Feature Request
Make exclusion of objects possible in the app.

### Problem Description
Sometimes, you see in the camera feed that a part of your multi-object print is failing, be it missing supports, be it adhesion issues. Continuing to print the failed part can result in spaghetti actually affecting even the intact parts. The fluidd UI gives you the possibility to exclude parts of the print from further processing (they need to be defined accordingly by the slicer, of course), however this functionality is missing in mobileraker.

### Proposed Solution
The "GCode preview" section should not only allow following the actual print, but also to exclude parts from printing.

### Alternatives Considered
Alternative is to use the web interface, which then immediately raises the question of why use the web interface and the app simultaneously in the first place.
### Additional Context

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [x] Provided a clear description of the problem you're facing.
- [x] Clearly outlined the proposed solution or feature.
- [x] Mentioned any alternative solutions or ideas you've considered.
- [x] Included any relevant context or supporting materials.

Clon1998, 2025-02-03:
This is already possible via the button right next to the current print field.

https://github.com/user-attachments/assets/c54c3514-7ee0-4bec-a1a2-689670c419ac

DBa20-24, 2025-02-03:
Indeed - could be documented more prominently somehow... Maybe a hint to this in the GCode preview section? (Because that's where fluidd provides this possibility)

Clon1998, 2025-02-03:
> Indeed - could be documented more prominently somehow... Maybe a hint to this in the GCode preview section? (Because that's where fluidd provides this possibility)

Probably,
I will see how I can make it more prominent.