Clon1998/mobileraker #470: Camera feed duplicated every time app gains focus

ColColonCleaner, 2025-02-03:
## Bug Report

### Description
The camera feed is duplicated every time the app app loses focus and regains it. This causes a lot of data transfer as camera feeds stack on top of each other. Only the latest feed is shown on the app, but all of them remain running.

### Steps to Reproduce
1. Have a way to see your bandwidth usage. On android you can use 'NetSpeed Indicator', it puts an indicator in your status bar.
2. Start Mobileraker l, viewing a printer with a camera. (I don't know if this bug is specific to mjpeg feeds, but that's what my USB webcam is set up as.)
3. Notice the net speed indicator shows your camera feed running, at whatever bandwidth is normal for your camera.
4. Switch to a different app for a moment, or just use the app drawer to look at your other running apps.
5. Switch back to mobileraker.
6. Notice your network bandwidth usage is now doubled since both camera feeds are running.
7. Switch to a different app again, then back to mobileraker. Notice the bandwidth usage is now 3x the original amount.

### Expected Behavior
The amount of bandwidth used by mobileraker does not multiply when switching out and back into it.

### Screenshots
https://github.com/user-attachments/assets/713d6683-3474-4be7-b589-83f4002ef6e7

### Version Information

- Device-OS (Android/iOS): Android 14
- Mobileraker: 2.8.4-588
- Klipper: v0.10.0-530-g3387a9c2-dirty
- Moonraker: v0.7.1-609-gbdd0222-dirty

### Debug Logs
GitHub's mobile app won't let me attach the logs.

Clon1998, 2025-02-03:
Hey,
are you using a pure MJPEG stream cam or a MJPEG Adaptive stream?

ColColonCleaner, 2025-02-03:
I believe it's a pure stream. I have the camera configured at a constant framerate rather than adaptive. Unless you mean something different by adaptive.

Clon1998, 2025-02-03:
> I believe it's a pure stream. I have the camera configured at a constant framerate rather than adaptive. Unless you mean something different by adaptive.

If you open up the printer settings in the app and go to the cam section, you can see if the cam type is `Adpative` or raw `Stream`.

ColColonCleaner, 2025-02-03:
It was set as MjpegStreamer instead of MjpegStreamerAdaptive.

I changed it to adaptive and that solved the bandwidth problem. Not sure why that worked, but it did! Thanks!

Clon1998, 2025-02-03:
> It was set as MjpegStreamer instead of MjpegStreamerAdaptive.
> 
> I changed it to adaptive and that solved the bandwidth problem. Not sure why that worked, but it did! Thanks!

Awesome!
However, the normal stream one still seems to keep the connection open. I will look into this!
Thanks!

commit 9686a234 (2025-02-04): fix: MjpegStream not closing old stream
fix #470

feat: Add apple IDFA
closed 2025-02-17 by Clon1998