Clon1998/mobileraker #471: Cant support the App and unlock all functionality

Roxxor91, 2025-02-05:
## Bug Report

### Description
Trying to support: Can not fetch supporter tiers!

### Steps to Reproduce
1. Go to 'Unterstützen Sie den Dev!'

### Expected Behavior
Be able to support the dev and unlock eg. Spoolman

### Version Information

- Device-OS (Android/iOS): LineageOS 22 (Android 15)
- Mobileraker: 2.8.4 from Github

### Additional Context
I do not use Gapps. I guess the support function relies on InApp PAyments thorugh Google Play. I obviously dont have that.

It would be nice if it would be possible to fully use the app and have a way to contribute without necessity of Google Apps.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.

point-source, 2025-02-08:
I similarly would like to support the app and unlock features but since I use GrapheneOS and have installed the app from the release page on this repo, it cannot reach Google Play to process payments. How else can I unlock the app?

Clon1998, 2025-02-09:
Please reach out via Mail to see what's possible.

dev@mobileraker.com