Clon1998/mobileraker #476: Mobile app notifications show a webcam error, despite when opening the app the webcam shows fine.

SethG911, 2025-02-08:
## Bug Report

### Description
Mobile app notifications show a webcam error, despite when opening the app the webcam shows fine.
### Steps to Reproduce
1. Start a print, or receive any status updates on print.
2. Receive start print notification on either iOS app or on Apple Watch.
3. Webcam snapshot shows a black error screen.
4. Open mobileraker app, webcam shows fine.

### Expected Behavior
The mobile app notification should show a snapshot of the webcam.

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information

- Device-OS: iOS
- Mobileraker: [app: 2.8.4, printer version: v0.4.0-91-g1d1e2ebe]
- Klipper: [v0.12.0-429-g01b0e98a]
- Moonraker: [v0.9.3-28-gf3b0afd5]

### Debug Logs
2025-02-08 10:30:25,906 mobileraker._Default INFO - Fetched 2 app Cfgs!
2025-02-08 10:30:25,907 mobileraker._Default INFO - Evaluate for machineID a21ae21f-033d-495f-a3b6-c0daf3f5fb39, cfg.version: 2.8.4-ios , cfg.snap: NotificationSnap(progress=0, progress_live_activity=0, progress_progressbar=0, last_progress=2025-02-08 10:15:46.963500, last_progress_live_activity=2025-01-16 18:32:09.652056, last_progress_progressbar=1970-01-01 00:00:00, state=printing, m117=c0f169737344e28e87eb123df627ba6ab840219985f7858a037cbef25b4e505e, gcode_response=f3ee8df62b3a71aeba521de4933acb3cb4277d549488a0fa8d7522088caa15a4, filament_sensors=[]), cfg.settings: NotificationSettings(progress_config=25, state_config=['standby', 'printing', 'paused', 'complete', 'error', 'cancelled'], android_progressbar=True, eta_sources=['filament', 'slicer', 'file'], created=2025-01-14T21:47:22.963729, last_modified=2025-01-15T09:49:43.662187)
2025-02-08 10:30:25,908 mobileraker._Default INFO - ProgressNoti preChecks: cfg.progress.config: 10 - 0 = 10 < 25 RESULT: False
2025-02-08 10:30:25,909 mobileraker._Default INFO - 0 Notifications for machineID: a21ae21f-033d-495f-a3b6-c0daf3f5fb39: state: False, proggress(text): False, M117 False, GcodeResponse: False, LiveActivity: False, FilamentSensor: 0, progressbar(android): False
2025-02-08 10:30:25,910 mobileraker._Default INFO - No snap update necessary for a21ae21f-033d-495f-a3b6-c0daf3f5fb39
2025-02-08 10:30:25,911 mobileraker._Default INFO - Evaluate for machineID 4eb5f4df-f278-463b-9bb1-1830e76cbc5e, cfg.version: 2.8.4-ios , cfg.snap: NotificationSnap(progress=0, progress_live_activity=5, progress_progressbar=0, last_progress=1970-01-01 00:00:00, last_progress_live_activity=2025-02-08 10:24:53.446477, last_progress_progressbar=1970-01-01 00:00:00, state=printing, m117=c0f169737344e28e87eb123df627ba6ab840219985f7858a037cbef25b4e505e, gcode_response=f3ee8df62b3a71aeba521de4933acb3cb4277d549488a0fa8d7522088caa15a4, filament_sensors=[]), cfg.settings: NotificationSettings(progress_config=-1, state_config=['error', 'complete', 'printing', 'paused'], android_progressbar=True, eta_sources=['slicer', 'file'], created=2025-01-23T10:21:49.351984, last_modified=2025-01-23T10:22:12.109574)
2025-02-08 10:30:25,912 mobileraker._Default INFO - LiveActivityUpdate preChecks passed
2025-02-08 10:30:25,913 mobileraker._Default INFO - LiveActivityUpdate passed. perc_interval_reached: True, time_interval_reached: False, eta_update: False
2025-02-08 10:30:25,914 mobileraker._Default INFO - LiveActivity (40552c2387fb319207c91d54880a49a26524581dc622af7ce6d7fc9857f6f8d98a9ad9e36d1786885a6379c1bb7204a442f67aa258f3735d96674ae3acb4d1e46bbf8b0b818bd8e83b5362e0f4749673):  10 - 1739035187
2025-02-08 10:30:25,915 mobileraker._Default INFO - 1 Notifications for machineID: 4eb5f4df-f278-463b-9bb1-1830e76cbc5e: state: False, proggress(text): False, M117 False, GcodeResponse: False, LiveActivity: True, FilamentSensor: 0, progressbar(android): False
2025-02-08 10:30:25,916 mobileraker._Default INFO - Updating snap in FCM Cfg for 4eb5f4df-f278-463b-9bb1-1830e76cbc5e: NotificationSnap(progress=0, progress_live_activity=10, progress_progressbar=0, last_progress=1970-01-01 00:00:00, last_progress_live_activity=2025-02-08 10:30:25.916565, last_progress_progressbar=1970-01-01 00:00:00, state=printing, m117=c0f169737344e28e87eb123df627ba6ab840219985f7858a037cbef25b4e505e, gcode_response=f3ee8df62b3a71aeba521de4933acb3cb4277d549488a0fa8d7522088caa15a4, filament_sensors=[])
2025-02-08 10:30:25,932 mobileraker._Default INFO - Updated snap in FCM Cfg for 4eb5f4df-f278-463b-9bb1-1830e76cbc5e: {'jsonrpc': '2.0', 'result': {'namespace': 'mobileraker', 'key': 'fcm.4eb5f4df-f278-463b-9bb1-1830e76cbc5e.snap', 'value': {'progress': 0.0, 'progress_live_activity': 0.1, 'progress_progressbar': 0.0, 'state': 'printing', 'm117': 'c0f169737344e28e87eb123df627ba6ab840219985f7858a037cbef25b4e505e', 'filament_sensors': [], 'gcode_response': 'f3ee8df62b3a71aeba521de4933acb3cb4277d549488a0fa8d7522088caa15a4', 'last_progress': '1970-01-01T00:00:00', 'last_progress_live_activity': '2025-02-08T10:30:25.916565', 'last_progress_progressbar': '1970-01-01T00:00:00'}}, 'id': 9811}
2025-02-08 10:30:25,934 mobileraker.cam INFO - Trying to take a snapshot from URI: http://127.0.0.1/webcam/?action=snapshot
2025-02-08 10:30:28,999 mobileraker.cam INFO - Took webcam snapshot! Rotating it using rotation 0°
2025-02-08 10:30:29,002 mobileraker.fcm INFO - Submitting 1 device-requets to mobileraker server
2025-02-08 10:30:29,900 mobileraker._Default INFO - ---- Completed Evaluations Task! ----
2025-02-08 10:35:26,761 mobileraker._Default INFO - Progress reached minimum interval. Evaluating!
2025-02-08 10:35:26,762 mobileraker._Default INFO - Snapshot passed threshold. LastSnap: PrinterSnapshot(timestamp=2025-02-08T10:30:25.896584,klippy_ready=True, print_state=printing, m117=Printing, m117_hash=c0f169737344e28e87eb123df627ba6ab840219985f7858a037cbef25b4e505e, virtual_sdcard=VirtualSDCard (progress: 0.101445, file_position: 845873), print_stats=PrintStats (filename: Neptune 4 UBBI reverse bowden v33_0.2mm_PLA_Elegoo Neptune 4 Plus_1h48m.gcode, total_duration: 879.0945845020001, print_duration: 847.9421545710002, state: printing, message: ), current_file=GCodeFile(filename=Neptune 4 UBBI reverse bowden v33_0.2mm_PLA_Elegoo Neptune 4 Plus_1h48m.gcode, modified=1739027603.804497, size=8338240, print_start_time=1739027667.1284664, job_id=000052, slicer=OrcaSlicer, slicer_version=2.2.0, gcode_start_byte=15362, gcode_end_byte=8319756, layer_count=100, object_height=None, estimated_time=6462, nozzle_diameter=0.4, layer_height=0.2, first_layer_height=0.28, first_layer_bed_temp=60.0, first_layer_extr_temp=210.0, chamber_temp=0.0, filament_name=Seth - Elegoo PLA Plus, filament_type=PLA, filament_total=5860.47, filament_weight_total=17.48), toolhead=<mobileraker.data.dtos.moonraker.printer_objects.Toolhead object at 0xffffa7479210>, gcode_move=<mobileraker.data.dtos.moonraker.printer_objects.GCodeMove object at 0xffffa7478150>, gcode_response=pressure_advance: 0.027250
// pressure_advance_smooth_time: 0.020000, gcode_response_hash=f3ee8df62b3a71aeba521de4933acb3cb4277d549488a0fa8d7522088caa15a4, timelapse_pause=None, filament_sensors=FilamentSensor(name=filament_sensor, enabled=True, filament_detected=True)), NewSnap: PrinterSnapshot(timestamp=2025-02-08T10:35:26.759875,klippy_ready=True, print_state=printing, m117=Printing, m117_hash=c0f169737344e28e87eb123df627ba6ab840219985f7858a037cbef25b4e505e, virtual_sdcard=VirtualSDCard (progress: 0.151329, file_position: 1261821), print_stats=PrintStats (filename: Neptune 4 UBBI reverse bowden v33_0.2mm_PLA_Elegoo Neptune 4 Plus_1h48m.gcode, total_duration: 1179.9564361040002, print_duration: 1148.8040061730003, state: printing, message: ), current_file=GCodeFile(filename=Neptune 4 UBBI reverse bowden v33_0.2mm_PLA_Elegoo Neptune 4 Plus_1h48m.gcode, modified=1739027603.804497, size=8338240, print_start_time=1739027667.1284664, job_id=000052, slicer=OrcaSlicer, slicer_version=2.2.0, gcode_start_byte=15362, gcode_end_byte=8319756, layer_count=100, object_height=None, estimated_time=6462, nozzle_diameter=0.4, layer_height=0.2, first_layer_height=0.28, first_layer_bed_temp=60.0, first_layer_extr_temp=210.0, chamber_temp=0.0, filament_name=Seth - Elegoo PLA Plus, filament_type=PLA, filament_total=5860.47, filament_weight_total=17.48), toolhead=<mobileraker.data.dtos.moonraker.printer_objects.Toolhead object at 0xffffa7269bd0>, gcode_move=<mobileraker.data.dtos.moonraker.printer_objects.GCodeMove object at 0xffffa726a110>, gcode_response=pressure_advance: 0.027250
[... cut ...]
snapshot_url = https://fluiddcam.sethgeib.com/?action=snapshot
# Flips the webcam image horizontally. Valid values are True or False
flip_horizontally = True
# Flips the webcam image vertically. Valid values are True or False
flip_vertically = True
# Rotates the webcam image. Valid values are 0, 90, 180, or 270
rotate = 0

[general]
# The temperature in Celsius that the bed must be under to be considered cooled down. This is used to fire the Bed Cooldown Complete notification.
bed_cooldown_threshold_temp_celsius = 40.0

Clon1998, 2025-02-08:
What error is shown in the screenshot/notification

SethG911, 2025-02-09:
<No Signal> Errors

[image]

[image]

SethG911, 2025-02-09:
When I click on the notification though, Mobileraker opens and the webcam shows fine.

[image]

SethG911, 2025-02-10:
Lord... I am so sorry for bugging you in creating this issue. I should have read the documentation more carefully. I added my reverse proxy to the snapshot_uri and it works perfectly! Thanks again for making such an amazing app, proud to be a supporter.

Clon1998, 2025-02-10:
> Lord... I am so sorry for bugging you in creating this issue. I should have read the documentation more carefully. I had no idea I could create that config file. I created it, added my reverse proxy address to the snapshot_uri, and it works perfectly! Thanks again for making such an amazing app, proud to be a supporter.

No worries!

closed 2025-02-10 by Clon1998