Clon1998/mobileraker #477: After deleting printer from app, I will get notifications.

Bomberski, 2025-02-08:
I have sold a printer and I'm getting notifications on android when they start prints. I also deleted the printer from the android app. How am I getting notifications when the printer is not even local?

## Bug Report

### Description
A clear and concise description of the bug.

### Steps to Reproduce
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See the error.

### Expected Behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information

- Device-OS (Android/iOS): [OS]
- Mobileraker: [version]
- Klipper: [version]
- Moonraker: [version]

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2025-02-08:
Hey,

the printer stores your phone's notifications ID in the moonraker database located on the host of the printer. This is by design and there is nothing that can be done remotely from my side to get the ID of the sold printer.

Therefore,
before selling the printer you should delete the moonraker database or clear the Notification cache.

Now you'll need to ask the new owner to clear the cache via the button at the very bottom of the printer edit view:

[image]

Clon1998, 2025-02-08:
Regarding the question why you get notifications of the printer is not local:

Notifications are not created by your phone but by the printer, the companion on the printer.

The companion uses the IDs stored in moonraker's database to submit notifications to my backend server which distributes them to either google or apples push service apis.

closed 2025-02-09 by Clon1998