Clon1998/mobileraker #480: Apple Watch Live Activity show incomplete status

b2un0, 2025-02-15:
## Bug Report

### Description
the status of the print is no longer displayed correctly on the apple watch live activity

### Expected Behavior
the live activity on apple watch should display the correct numbers or times

### Screenshots

[image]
[image]
[image]
[image]

### Version Information

- Device-OS: iPhone 13 mini iOS 18.3.1 and Apple Watch Series 10 watchOS 11.3.1
- Mobileraker App: 2.8.4 (on booth printers)
- Mobileraker integration:  v0.4.0-91-g1d1e2ebe (on both printers)
- Klipper: v0.12.0-432-gfec3e685 (on both printers)
- Moonraker: v0.9.3-48-g354cc11b (on both printers)

### Debug Logs

[mobileraker_2025-02-15T09-42-08.log](https://github.com/user-attachments/files/18811754/mobileraker_2025-02-15T09-42-08.log)
[mobileraker_2025-02-15T17-03-37.log](https://github.com/user-attachments/files/18811752/mobileraker_2025-02-15T17-03-37.log)
[mobileraker_2025-02-15T12-09-46.log](https://github.com/user-attachments/files/18811753/mobileraker_2025-02-15T12-09-46.log)
[mobileraker_2025-02-15T15-39-53.log](https://github.com/user-attachments/files/18811756/mobileraker_2025-02-15T15-39-53.log)
[mobileraker_2025-02-15T13-17-02.log](https://github.com/user-attachments/files/18811755/mobileraker_2025-02-15T13-17-02.log)

### Additional Context
I already noticed this a few weeks ago

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.
- [x] Attached Mobileraker's debug log files.
- [x] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2025-02-16:
Thanks for letting me know.
I assume the latest watch OS update broke the UI.

It might take some time to get this fixed because I am rather busy currently.
In general I am not really happy with the live activities and they need another rework from my end.

- Patrick

Clon1998, 2025-02-18:
Seems like I will need to adjust the activity to offer a "SmartStack" specific one that works on [WatchOs.11](https://developer.apple.com/design/human-interface-guidelines/live-activities#watchOS).

Clon1998, 2025-03-01:
Hey,

I am currently experimenting with the WatchOS live activity design and Id appreciate some feedback.

Would that design work or is a more basic version without the logo better?

I tried out both versions today on my watch and without the logo I was first a bit confused to what the watch was showing.

[image]

[image]

b2un0, 2025-03-02:
Definitely with the logo

b2un0, 2025-03-02:
Is it possible to Show the thumbnail from the gcode as background Image?

closed 2025-03-05 by Clon1998

Clon1998, 2025-03-05:
> Is it possible to Show the thumbnail from the gcode as background Image?

I will need to look into that. It might be possible but for now I will just use the logo.

The thumbnail highly depends on the slicer. Some just show the to-print object. While others also include the "printer's bed" and so on.

For now I'll stick with this, which hopefully should start rolling out tomorrow:

[image]

[image]

commit 01a054d2 (2025-02-28): feat: Add support for WatchOS smart stack

fix #480
commit c32e3cf6 (2025-03-01): feat: Adjust styling

fix #480