Clon1998/mobileraker #483: Option to disable printers in app

Peaches491, 2025-02-18:
## Feature Request
Enable/Disable option for each printer in app

### Problem Description
I have separate profiels for Ethernet and Wi-Fi connections to my printer, e.g. `My Printer` and `My Printer Wi-Fi`. Normally my printer is stationary, and connected to ethernet. However, some times I need to move it to another room for a long print, since I work from home. 

Since there is no option to disable a printer, I get double notifications for every print status change, and I assume the app is hitting the printer API twice as frequently

### Proposed Solution
I'd like an option to _disable_ `My Printer Wi-Fi` when not in use (i.e. when plugged in to ethernet), and then re-enable it whenever my pritner is "mobile"

I think a disabled printer should:
- Make no connection attempts
- Send no notifications
- Have all the buttons greyed out (or make it not selectable in the "Manage Printers" dropdown)

### Alternatives Considered
The alternative would be to delete the Wi-Fi configration and then re-add it every time I  need to access the printer via Wi-Fi

### Additional Context
Add any screenshots, examples, or additional context to help us understand your feature request better.

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [x] Provided a clear description of the problem you're facing.
- [x] Clearly outlined the proposed solution or feature.
- [x] Mentioned any alternative solutions or ideas you've considered.
- [x] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Clon1998, 2025-02-18:
I think you can make use of the "Remote Connection" Manual setting?

Just configure your Eth. as default and the Wifi as "Manual" setting with an alternative add and headers.
The app will first try to use the Eth add and if its not reachable it tries the remote one.

Peaches491, 2025-02-18:
@Clon1998 I believe that setting is for when *I* am not on the same network. 

In the scenario I described, my phone and I are stationary, and always connected to the same network as the printer. The printer just gets a different IP when it's in Wi-Fi. 

I'll update the dsecription to clear that up

Clon1998, 2025-02-18:
> [@Clon1998](https://github.com/Clon1998) I believe that setting is for when _I_ am not on the same network.
>
> In the scenario I described, my phone and I are stationary, and always connected to the [...]

Yes I got that. However, if the ETH Ip is not reachable due the printer not being connected to it/using wifi. Connecting to the ETH Ip will cause a connection error which itself will trigger a HandOver to the "Remote Connection", in your case WiFi Ip.

Peaches491, 2025-02-20:
Ah, I've tried your suggestion @Clon1998 and it does what you describe. After 10 seconds, it tries the fallback IP and finds the printer. Thanks for the tip!

The 10 second delay is almost a deal breaker however, because I have to wait for the wired connection to timeout nearly _every time_ I open the app. I think this feature is better suited to SSID detection, where it can detect if you're home and use your local IP, or otherwise access via some external IP or VPN

I'd still like the disable option for a second reason:
My friend has a 3D printer which he's given me access to as well through the app. I don't want to get notifications every time he starts/stops a print either, I'd like to disable his printer until I specifically want to work with it.

Clon1998, 2025-03-05:
> Ah, I've tried your suggestion [@Clon1998](https://github.com/Clon1998) and it does what you describe. After 10 seconds, it tries the fallback IP and finds the printer. Thanks for the tip!
>
> The 10 [...]

Alright,
I will look into that :)