Clon1998/mobileraker #484: Ability to disable notifications from sensors

ksummers92, 2025-02-20:
## Feature Request

### Problem Description
Using ERCF V2 with toolhead, extruder, and feedback sensors means that I get a lot of notifications when these sensors change state. For the feedback sensor, it is designed to trigger as part of normal operation so the notifications are not helpful. 

### Proposed Solution
An option to disable notifications for individual sensors or, at a minimum, the ability to disable notifications for sensors in general. 

### Alternatives Considered
Currently, I have just disabled notifications for the entire app in the iOS settings because I would get 100+ notifications every print. 

### Additional Context
Add any screenshots, examples, or additional context to help us understand your feature request better.

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [x] Provided a clear description of the problem you're facing.
- [x] Clearly outlined the proposed solution or feature.
- [x] Mentioned any alternative solutions or ideas you've considered.
- [x] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Clon1998, 2025-02-20:
This is already possible.
Not via the app but via the companion [config](https://github.com/Clon1998/mobileraker_companion?tab=readme-ov-file#companion---config).

Just add a `ignore_filament_sensors` with a list of your sensors

ksummers92, 2025-02-20:
Ah! My apologies! I had no idea that the companion config was a thing! I always assumed that everything was configured in the app and I didn't see an option in the app. Thank you for the clarification!

closed 2025-02-20 by ksummers92