Clon1998/mobileraker #486: Klipper error after update

slayeralive, 2025-03-03:
## Bug Report

### Description
A clear and concise description of the bug.

### Steps to Reproduce
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See the error.

### Expected Behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain the problem.

### Version Information

- Device-OS (Android/iOS): [OS]
- Mobileraker: [version]
- Klipper: [version]
- Moonraker: [version]

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

[image]

Clon1998, 2025-03-03:
Please attach the klipper + moonraker version

slayeralive, 2025-03-03:
> Please attach the klipper + moonraker version

[image]

slayeralive, 2025-03-03:
[image]

Clon1998, 2025-03-03:
Can't reproduce it and the configuration is valid. The mentioned non parsable fields are also valid moonraker fields (See [docs](https://moonraker.readthedocs.io/en/latest/configuration/#git-repo-configuration)) 

I suggest you restart moonraker once your current print is done. If the warning message persist you might want to verify that the mobileraker update section is actually part of moonraker's config and not the klipper config file.

LanceChronos, 2025-04-18:
[image]

I had a similar warning for KAMP, and the update manager is missing from my machine tab.

Clon1998, 2025-04-19:
> [image]
> 
> I had a similar warning for KAMP, and the update manager is missing from my machine tab. 

none of these errors mentions mobileraker.

You have 
- one faulty config for sonar.
- insufficient storage for moonraker's history api

closed 2025-05-22 by Clon1998