Clon1998/mobileraker #487: Webrtc scaling? issue

m-klecka, 2025-03-05:
## Bug Report

### Description
When using webrtc for the Webcam the image looks like it has some scaling problems with weird image. When using mjoeg it's fine. Also when viewing the same webrtc stream in Mainsail, the stream looks also fine. 

### Steps to Reproduce
1. Use webrtc camera stream 

### Expected Behavior
Look the same as in mainsail

### Screenshots
![Screenshot_2025-03-05-16-03-52-41_424ec53f7d515395b4c3ef1aaa0a00a2.jpg](https://github.[image]

_2025-03-05-16-06-01-18_3aea4af51f236e4932235fdada7d1643.jpg …]()

[image]

### Version Information

- Device-OS (Android/iOS): Android
- Mobileraker: 2.8.6-615
- Klipper: 0.12.0-439-g1fc6d214f
- Moonraker: v0.9.3-59-g62051108

### Debug Logs

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [✅] Provided a clear bug description.
- [✅] Listed detailed steps to reproduce the issue.
- [✅] Described the expected behavior.
- [✅] Included the Mobileraker version you are using.
- [✅] Attached Mobileraker's debug log files.
- [✅] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2025-03-06:
Can you share you crownnest logs?

m-klecka, 2025-03-14:
> Can you share you crownnest logs?

Sure, here.

[crowsnest.log](https://github.com/user-attachments/files/19248758/crowsnest.log)

Clon1998, 2025-03-22:
Hey,
Sadly, I was not able to reproduce that issue. Can you do me a favor and check if the same issues persist with the different resolutions and also if this only happens for the webcam card or the fullscreen cam?

Your other supported resolutions are:
```
1920x1080
640x680
800x600
1024x768
1280x720
1280x1024
320x240
```