Clon1998/mobileraker #488: Unusable with MicroG

mawoka-myblock, 2025-03-05:
## Bug Report

### Description
A clear and concise description of the bug.

### Steps to Reproduce
1. Install the app using the Aurora Store on an Android ROM without Google Play Services, but with MicroG.

### Expected Behavior
App should work

### Screenshots

[image]

### Version Information

- Device-OS (Android/iOS): Android 14 with MicroG Version 0.3.2.240913
- Mobileraker: 2.8.6
- Klipper: Not applicable
- Moonraker: Not applicable

### Debug Logs
Doesn't matter

### Additional Context
- It worked in the past
- Possible solution: Provide build not depending on Google Play Services

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [x] Provided a clear bug description.
- [x] Listed detailed steps to reproduce the issue.
- [x] Described the expected behavior.
- [x] Included the Mobileraker version you are using.
- [x] Attached Mobileraker's debug log files.
- [x] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2025-03-22:
Hey,
can you check for me if 2.8.7 still has the same problem?

mawoka-myblock, 2025-03-22:
Yea, unfortunately yes...

nagyrobi, 2025-04-12:
Same here...
Please remove google dependency...

blumberg, 2026-08-25:
I'm using GrapheneOS Android version 17, build 2026081300, the app is working fine for me now.
If someone else can confirm, I believe this issue can be closed