Clon1998/mobileraker #493: Possible errantly repeated print

ffolkes, 2025-03-26:
I'm not sure if this is a Mobileraker bug, but I just want to mention it.

The day before yesterday I needed to reprint something, so I opened the app, and was easily able to start the print. It completed, and that was that. Then...10 minutes later I heard slipping stepper motors and cracking plastic. I rushed over and yanked the plug. For reasons unknown, 10 minutes after the first job, something told klipper to start the print again. With the completed part on the bed, it tried to autohome and thus destroyed parts of my fan duct/bracket.

This, nor anything like it, has never happened in 6 years of 3D printing. The only thing different was that I used Mobileraker to start the print job, for the first time ever. Until now, I had only used it to check on the status of prints.

I'm using the latest version of Mobileraker on Android, and latest klipper and moonraker. I update those daily, so maybe this was a bug that was fixed yesterday/today.

Just wanted to let you know. Thanks for making this app, it looks beautiful!

Clon1998, 2025-03-26:
Sorry to hear about that.
I will verify if that issue was caused by Mobileraker but you're the first one to ever report that behavior.

Nerveless I will double check it.

Clon1998, 2025-03-26:
Can you do me a favor and attach the debug logs?
Setting page-> Scroll to bottom -> Debug logs

ffolkes, 2025-03-26:
Thank you for your attention, I doubt this has anything to do with Mobileraker. I see you include several logs, which is very handy. The log for that night (attached) does not show a second print. I'll try to reproduce this later, but I think it was likely a freak accident of some sort. Thank you anyway!

[mobileraker_2025-03-25T04-06-43.log](https://github.com/user-attachments/files/19476023/mobileraker_2025-03-25T04-06-43.log)

Clon1998, 2025-03-30:
Hey,
I just checked your logs and just a single print job was started:

```log
04:07:22.513 (+0:00:39.335354)
💡 Starting print for file: CE3PRO_PlantHooksB_r6_newererer.gcode
```

I also checked my codebase and I dont see anything that could have caused that.

Therefore, I will close this issue.

closed 2025-03-30 by Clon1998