Clon1998/mobileraker #496: No error notifications being sent

BlackNet, 2025-04-03:
## Bug Report

### Description
A clear and concise description of the bug.

When klipper enters an error state no notifications are being sent out.  I DO get online, pause, printing, canceled, etc work but nothing with error.

Clon1998, 2025-04-03:
Please share the app logs alongside the logs of the companion.

BlackNet, 2025-04-03:
this is klippy and mobileraker logs.

[logs-20250403-092304.zip](https://github.com/user-attachments/files/19587976/logs-20250403-092304.zip)

I do not have the heater plugged in, I set the heater to 50c to force it klipper to fail.  Klipper does error out with this.    

[image]

Just no alerts via mobileraker.  on the app 

[image]
[image]

BlackNet, 2025-04-03:
Yes. In the voron community, we commonly will heat soak the printer.  This could be idle for an hour or longer, waiting for the frame to heat up.   Some will even turn on the bed and let it soak until they are ready to print.

Clon1998, 2025-04-07:
> Yes. In the voron community, we commonly will heat soak the printer before printing.  This could be idle for an hour or longer, waiting for the frame to heat up.   Some will even turn on the bed and let it soak until they are ready to print.    

I get that, but why would it be useful to then?
In case klipper throws an error while heating up?

BlackNet, 2025-04-07:
Yes. An error is an error. If given the options for mine, I would tell it to give me error codes at any time, regardless of print status. Only showing errors while printing kind of seems frugal.

Many errors could be thrown while heat soaking. Unexpected rate is a common one.