Clon1998/mobileraker #501: Show webcam even when there is an error

rcambrj, 2025-04-06:
## Feature Request

### Problem Description
When klipper is in an error state, for example, when the MCU is disconnected, mobileraker shows an error screen and does not display the webcam(s).

In the case of an emergency, shutting down the MCU is paramount, but one might still want to see what's happening.

### Proposed Solution
Always display the webcam feed(s) if configured, even in an error state.

### Alternatives Considered
For now, bookmarking the webcam feed in a browser is a workaround.

### Additional Context
None

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [ ] Provided a clear description of the problem you're facing.
- [ ] Clearly outlined the proposed solution or feature.
- [ ] Mentioned any alternative solutions or ideas you've considered.
- [ ] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Clon1998, 2025-04-07:
Hey,
AS far as I am aware the app is already doing that or am I missing something?

[image]

rcambrj, 2025-04-07:
Thanks for the quick response!

Not for me:

[image]

Looks like you're using iOS. Maybe this is exclusive to Android?

Clon1998, 2025-04-07:
> Thanks for the quick response!
> 
> Not for me:
> 
> [image]
> 
> Looks like you're using iOS. Maybe this is exclusive to Android?

Can you provide me the version number installed?
Either available at the bottom of the settings page or at the top of the changelog page.

rcambrj, 2025-04-07:
2.8.7-618

rcambrj, 2025-04-07:
I'm so sorry to have wasted your time. I recently reset moonraker, so the webcam configuration went away. Configuring the webcam fixed the problem. Again I'm sorry but thanks so much for jumping on this so quickly!

closed 2025-04-07 by rcambrj