Clon1998/mobileraker #502: Snapshot Notifications

ddtav478, 2025-04-09:
## Feature Request

### Problem Description
Notifications do not include Snapshot from Camera. Currently on iOS, push notifications are received, however, an image is not present with the notification regardless of settings in mobileraker.conf 

### Proposed Solution
A method to include a snapshot from the camera with push notifications

### Alternatives Considered
Maybe a separate notification would help.

### Additional Context

[image]

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [ ] Provided a clear description of the problem you're facing.
- [ ] Clearly outlined the proposed solution or feature.
- [ ] Mentioned any alternative solutions or ideas you've considered.
- [ ] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Clon1998, 2025-04-09:
Hey,
this is working as expected as long as you're not a supporter since notifications with snapshots are exclusive to supporters.

ddtav478, 2025-04-09:
Thanks for the response. 

I am a lifetime supporter at this time. Not sure if my settings for my phone, printer, or app are incorrect.

Clon1998, 2025-04-09:
> Thanks for the response.
> 
> I am a lifetime supporter at this time. Not sure if my settings for my phone, printer, or app are incorrect.

Gotcha,
can you upload the app log files located on the bottom of the app settings page alongside the companion log files to me?

ddtav478, 2025-04-10:
Found the issue. The mobileraker.conf contained the entire line of code even with the '#' (posted below). Silly mistake on my part. You're correct in stating that it works as advertised. Will be closing the thread. Thank you. 

'/webcam/?action=snapshot #http://127.0.0.1/webcam/?action=snapshot'

closed 2025-04-10 by ddtav478