Clon1998/mobileraker #504: New files doesn't show up in app till I print them.

Johnnytk36, 2025-04-12:
## Bug Report

### Description
I can not see new files in my files list from the mobileraker app unless I print it. Once I start printing them, they show up in the file list. This has been happening since an update or two ago. I cant remember when it started.

I can see the new file in the file list when i view the printer from the web interface. 

### Screenshots

[image]
[image]

### Version Information

- Device-OS (Android/iOS): iOS
- Mobileraker: 2.8.7
- Klipper: Sonic Pad

### Debug Logs

[mobileraker_2025-04-09T21-53-58.log](https://github.com/user-attachments/files/19715281/mobileraker_2025-04-09T21-53-58.log)
[mobileraker_2025-04-10T18-00-22.log](https://github.com/user-attachments/files/19715279/mobileraker_2025-04-10T18-00-22.log)
[mobileraker_2025-04-11T12-19-23.log](https://github.com/user-attachments/files/19715282/mobileraker_2025-04-11T12-19-23.log)
[mobileraker_2025-04-11T12-31-43.log](https://github.com/user-attachments/files/19715280/mobileraker_2025-04-11T12-31-43.log)
[mobileraker_2025-04-11T18-58-11.log](https://github.com/user-attachments/files/19715278/mobileraker_2025-04-11T18-58-11.log)

Clon1998, 2025-04-12:
Hey,
I from the screenshots I can see that you currently sort by the last printed date. New files don't have such a date and therefore don't show up in the list.
This actually is an UX error on my end all files should always show up no matter if they were printed or not.

Can you confirm to me that if you switch the sorting to last modified, the missing file appears?

Johnnytk36, 2025-04-12:
Well. I can't believe i didn't see that myself.....

I uploaded a test file and it didnt show, but as soon as i said sort by name it showed up. I went back to sort by last printed, and it was still there without printing it. 

I tried uploading some more test files and they are showing up even when i have it set to "last printed" without me changing the sort, or printing them. Weird.

Clon1998, 2025-04-12:
Interesting, that's definitely not how it's supposed to work.
I will try to reproduce this behavior.

In addition, I will make sure to include all files in all sort modes at all time

Clon1998, 2025-04-12:
https://github.com/user-attachments/assets/a814098a-53c3-45b4-ac2f-6efc270dbe96

Actually, I already included all files and it also always shows up?

Johnnytk36, 2025-04-12:
I wonder if mine was stuck and changing the sort mode fixed it. Like i mentioned, since i changed the sort method once, i cant reproduce the error again now no matter what sort mode  am in. 

Was it because i had this set like this through some updates. It was months ago i changed the sort mode i bet.

Johnnytk36, 2025-04-12:
Actually, take a look at this video. This is another printer i have. I cant get it to show up when they are sorted by name or last printed. Not until i sort by last modified do they show up. You can see the green bar when it detects that i sent the file from my laptop via prusaslier. 

[Google Drive Video](https://drive.google.com/file/d/1LxRokkvhSLtMaumawD15TW1D3pjqrf7t/view?usp=drive_link)

Clon1998, 2025-04-12:
Mhh,
I just tried to reproduce it by uploading files directly from the slicer (Orca in my case)
But the files always show up no matter if I am sorting by name, size, modified, printed...

Can you do me a favor, and re upload the log files in the hope that the one from the recording is still available.

Maybe I can find something interesting there.

Johnnytk36, 2025-04-12:
Here you go. I did it one more time with that printer.

This is a voron 2.4 that is running on a CB1 with klipper version v0.12.0-404 

[mobileraker_2025-04-11T12-31-43.log](https://github.com/user-attachments/files/19720366/mobileraker_2025-04-11T12-31-43.log)
[mobileraker_2025-04-11T12-19-23.log](https://github.com/user-attachments/files/19720367/mobileraker_2025-04-11T12-19-23.log)
[mobileraker_2025-04-11T18-58-11.log](https://github.com/user-attachments/files/19720364/mobileraker_2025-04-11T18-58-11.log)
[mobileraker_2025-04-10T18-00-22.log](https://github.com/user-attachments/files/19720368/mobileraker_2025-04-10T18-00-22.log)
[mobileraker_2025-04-12T01-19-30.log](https://github.com/user-attachments/files/19720365/mobileraker_2025-04-12T01-19-30.log)

Clon1998, 2025-04-13:
I reviewed the logs, and everything appears to be functioning properly. I tested several different scenarios today and could not replicate the issue on any of my three printers.

I will continue to monitor for that problem. If you find a reliable way to reproduce it, I would appreciate a detailed description. Additionally, could you check the search function to determine if it is absent in the normal user interface, or if it can still be accessed through the search feature?

Johnnytk36, 2025-04-14:
Sounds good. Thank you for looking. I will try and spend some time earlier this week to see if I can find a way for you to reproduce it.

Clon1998, 2025-05-22:
Hey,
just wanted to let you know that I identified the and fixed the bug.

The primary reason was a calculation issue on my end due to index shifting of the file browser view.

The fix will be included in the upcoming release.

commit d280a8e2 (2025-05-19): fix: Adjust ad display logic in file manager for correct item indexing

fixes #504
closed 2025-05-22 by Clon1998