Clon1998/mobileraker #505: Mobileraker won’t connect to Kobra S1

dJOS1475, 2025-04-12:
Howdy, first up, I love your app and use it on all my printers. However, after jail breaking my new Kobra S1, I found that Mobileraker won’t connect to it.

I’ve already discussed this with the Rinkhals dev and the info is linked below.

https://github.com/jbatonnet/Rinkhals/issues/133#issuecomment-2799028115

`exception:%0A%20type%20'Null'%20is%20not%20a%20subtype%20of%20type%20'num'%20in%20type%20cast%0A%0A%230%20%20%20%20%20%20_$$ConfigStepperImplFromJson%20(package:common/data/dto/config/config_stepper.g.dart:17)%0A%231%20%20%20%20%20%20new%20_$ConfigStepperImpl.fromJson%20(package:common/data/dto/config/config_stepper.freezed.dart:346)%0A%232%20%20%20%20%20%20_$ConfigStepperFromJson%20(package:common/data/dto/config/config_stepper.freezed.dart:18)%0A%233%20%20%20%20%20%20new%20ConfigStepper.fromJson%20(package:common/data/dto/config/config_stepper.dart:40)%0A%234%20%20%20%20%20%20new%20ConfigFile.parse%20(package:common/data/dto/config/config_file.dart:77)%0A%235%20%20%20%20%20%20PrinterBuilder._updateConfigFile%20(package:common/data/dto/machine/printer_builder.dart:250)%0A%236%20%20%20%20%20%20PrinterBuilder.partialUpdateField%20(package:common/data/dto/machine/printer_builder.dart:220)%0A%237%20%20%20%20%20%20PrinterService._parseObjectType%20(package:common/service/moonraker/printer_service.dart:560)%0A%238%20%20%20%20%20%20PrinterService._temperatureStore.%3Canonymous%20closure%3E%20(package:common/service/moonraker/printer_service.dart:514)%0A%239%20%20%20%20%20%20_LinkedHashMapMixin.forEach%20(dart:_compact_hash:726)%0A%2310%20%20%20%20%20PrinterService._parseQueriedObjects%20(package:common/service/moonraker/printer_service.dart:575)%0A%2311%20%20%20%20%20PrinterService._printerObjectsQuery%20(package:common/service/moonraker/printer_service.dart:599)%0A%3Casynchronous%20suspension%3E%0A%2312%20%20%20%20%20PrinterService.refreshPrinter%20(package:common/service/moonraker/printer_service.dart:183)%0A%3Casynchronous%20suspension%3E%0A%0A%0AFailed-Key:%20configfile%20%0ARaw%20Json:%0A%7B%22configfile%22:%7B%22config%22:%7B%22bed_mesh%22:%7B%7D,%22buttons%22:%7B%7D,%22controller_fan%20controller_fan%22:%7B%7D,%22cs1237%22:%7B%7D,%22display_status%22:%7B%7D,%22engine_mode%22:%7B%7D,%22exclude_object%22:%7B%7D,%22extruder%22:%7B%7D,%22fan%22:%7B%7D,%22fan_generic%20air_filter_fan%22:%7B%7D,%22fan_generic%20box_fan%22:%7B%7D,%22filament_hub%22:%7B%7D,%22filament_tracker%22:%7B%7D,%22flow_calibration%22:%7B%7D,%22force_move%22:%7B%7D,%22gcode_arcs%22:%7B%7D,%22gcode_macro%20CANCEL_PRINT%22:%7B%7D,%22gcode_macro%20WIPE_ENTER%22:%7B%7D,%22gcode_macro%20WIPE_EXIT%22:%7B%7D,%22gcode_macro%20WIPE_NOZZLE%22:%7B%7D,%22gcode_macro%20WIPE_NOZZLE1%22:%7B%7D,%22gcode_macro%20WIPE_STOP%22:%7B%7D,%22gcode_macro%20WIPE_STOP1%22:%7B%7D,%22heater_bed%22:%7B%7D,%22heater_fan%20extruder_fan%22:%7B%7D,%22input_shaper%22:%7B%7D,%22led_pin%22:%7B%7D,%22leviQ3%22:%7B%7D,%22lis2dw12%22:%7B%7D,%22mcu%22:%7B%7D,%22mcu%20nozzle_mcu%22:%7B%7D,%22ota_filament_hub%22:%7B%7D,%22pause_resume%22:%7B%7D,%22pid_calibrate%22:%7B%7D,%22printer%22:%7B%7D,%22printer_marco%22:%7B%7D,%22probe%22:%7B%7D,%22resonance_tester%22:%7B%7D,%22safe_z_home%22:%7B%7D,%22stepper_enable%22:%7B%7D,%22stepper_x%22:%7B%7D,%22stepper_y%22:%7B%7D,%22stepper_z%22:%7B%7D,%22timelapse%22:%7B%7D,%22tmc2209%20extruder%22:%7B%7D,%22tmc2209%20stepper_x%22:%7B%7D,%22tmc2209%20stepper_y%22:%7B%7D,%22tmc2209%20stepper_z%22:%7B%7D,%22verify_heater%20extruder%22:%7B%7D,%22verify_heater%20heater_bed%22:%7B%7D,%22virtual_sdcard%22:%7B%7D,%22gcode_macro%20pause%22:%7B%7D,%22gcode_macro%20resume%22:%7B%7D,%22gcode_macro%20cancel_print%22:%7B%7D%7D,%22save_config_pending%22:true,%22save_config_pending_items%22:%7B%22bed_mesh%20default%22:%7B%22algo%22:%22lagrange%22,%22max_x%22:%22245%22,%22max_y%22:%22245%22,%22mesh_x_pps%22:%222%22,%22mesh_y_pps%22:%222%22,%22min_x%22:%225%22,%22min_y%22:%225%22,%22points%22:%22-0.757167,%20-0.551667,%20-0.360500,%20-0.295667,%20-0.182167%5Cn-0.572500,%20-0.342000,%20-0.070167,%20-0.076667,%20-0.057000%5Cn-0.491833,%20-0.204500,%200.028500,%20-0.011500,%20-0.110667%5Cn-0.593333,%20-0.406667,%20-0.180333,%20-0.256000,%20-0.413167%5Cn-0.770833,%20-0.615667,%20-0.462500,%20-0.467000,%20-0.733833%22,%22tension%22:%220.2%22,%22version%22:%221%22,%22x_count%22:%225%22,%22y_count%22:%225%22%7D,%22leviQ3%22:%7B%22auto_zoffset_on_off%22:%221%22,%22bed_temp_cfg%22:%2255.000000%22%7D,%22probe%22:%7B%22z_offset%22:%22-0.0975%22%7D%7D,%22settings%22:%7B%22bed_mesh%22:%7B%22algorithm%22:%22lagrange%22,%22bicubic_tension%22:0.2,%22fade_end%22:0,%22fade_start%22:1,%22fade_target%22:0,%22horizontal_move_z%22:3,%22mesh_radius%22:0,%22move_check_distance%22:5,%22speed%22:300,%22split_delta_z%22:0.0125%7D,%22controller_fan%20controller_fan%22:%7B%22cycle_time%22:0.01,%22fan_speed%22:1,%22heater%22:%5B%22extruder%22%5D,%22idle_speed%22:0,%22kick_start_time%22:0.5,%22max_power%22:1,%22off_below%22:0,%22pin%22:%22PB5%22,%22shutdown_speed%22:0,%22stepper%22:%5B%5D%7D,%22cs1237%22:%7B%22dout_pin%22:%22nozzle_mcu:PA6%22,%22level_pin%22:%22nozzle_mcu:PA7%22,%22sclk_pin%22:%22nozzle_mcu:PA5%22%7D,%22extruder%22:%7B%22control%22:%22pid%22,%22dir_pin%22:%22!nozzle_mcu:PA11%22,%22enable_pin%22:%22!nozzle_mcu:PC13%22,%22filament_diameter%22:1.75,%22gear_ratio%22:%5B%5D,%22heater_pin%22:%22nozzle_mcu:PB8%22,%22inline_resistor%22:0,%22instantaneous_corner_velocity%22:1,%22max_extrude_cross_section%22:50,%22max_extrude_only_accel%22:5321.6216073828855,%22max_extrude_only_distance%22:100,%22max_extrude_only_velocity%22:159.64864822148658,%22max_power%22:1,%22max_safe_temp%22:320,%22max_temp%22:340,%22min_extrude_temp%22:170,%22min_safe_temp%22:170,%22min_temp%22:-180,%22nozzle_diameter%22:0.4,%22nozzle_material%22:%22brass%22,%22pid_kd%22:108.268,%22pid_ki%22:0.698,%22pid_kp%22:17.393,%22pressure_advance%22:0.03,%22pressure_advance_max%22:0.1,%22pressure_advance_smooth_time%22:0.04,%22pullup_resistor%22:4700,%22pwm_cycle_time%22:0.1,%22rotation_distance%22:6.5756,%22sensor_pin%22:%22nozzle_mcu:PA0%22,%22sensor_type%22:%22NTC%20ZM3486%20HNS112HC4570G0060Y3486%22,%22smooth_time%22:1,%22step_pin%22:%22nozzle_mcu:PA12%22%7D,%22fan%22:%7B%22cycle_time%22:0.005,%22kick_start_time%22:0.5,%22max_power%22:1,%22off_below%22:0,%22pin%22:%22nozzle_mcu:PB6%22,%22shutdown_speed%22:0%7D,%22fan_generic%20air_filter_fan%22:%7B%22cycle_time%22:0.00003333333,%22kick_start_time%22:0.5,%22max_power%22:1,%22off_below%22:0,%22pin%22:%22PC7%22,%22shutdown_speed%22:0%7D,%22fan_generic%20box_fan%22:%7B%22cycle_time%22:0.00003333333,%22kick_start_time%22:0.5,%22max_power%22:1,%22off_below%22:0,%22pin%22:%22PC6%22,%22shutdown_speed%22:0%7D,%22filament_hub%22:%7B%22action_timeout%22:240,%22config_file%22:%22/userdata/app/gk/config/ams_config.cfg%22,%22cruising_speed%22:200,%22cutter_pin%22:%22nozzle_mcu:PB1%22,%22cutter_position%22:1.5,%22default_feed_speed%22:30,%22default_unwind_speed%22:20,%22delay_before_throw%22:3,%22feed_max_length%22:2600,%22flush_multiplier%22:1.5,%22impact_position_x%22:260,%22impact_position_y%22:25,%22impact_speed%22:10,%22pre_sweep_position_x%22:47,%22pre_sweep_position_y%22:230,%22recalibrate_period%22:7200,%22serial%22:%22%22,%22serial1%22:%22%22,%22serial_prefix%22:%22usb-ANYCUBIC_ACE%22,%22shake_filament_speed%22:100,%22sweep_fan_speed%22:0,%22sweep_position_x%22:47,%22sweep_position_y%22:276,%22sweep_reprint_extrude_len%22:0.9,%22sweep_retract_len%22:0.4,%22sweep_speed%22:10,%22switch_pin%22:%22nozzle_mcu:PB0%22,%22throw_impact_speed%22:200,%22throw_position_x%22:47,%22throw_position_y%22:276,%22unwind_length_after_triggered%22:1100,%22unwind_speed_old_ace%22:15%7D,%22filament_tracker%22:%7B%22length_per_pulse%22:1.86532063806894,%22safe_unwind_len%22:150,%22signal_type%22:%22adc%22,%22tracker_detect_pin%22:%22PB0%22,%22tracker_encoder_pin%22:%22PB1%22%7D,%22flow_calibration%22:%7B%22delta_press_max%22:60000,%22fit_val_max%22:20000%7D,%22gcode_arcs%22:%7B%22resolution%22:1%7D,%22gcode_macro%20cancel_print%22:%7B%22description%22:%22G-Code%20macro%22,%22gcode%22:%22CANCEL_PRINT_BASE%22,%22rename_existing%22:%22CANCEL_PRINT_BASE%22%7D,%22gcode_macro%20wipe_enter%22:%7B%22description%22:%22G-Code%20macro%22,%22gcode%22:%22G90%5CnG1%20F36000%5CnG1%20Y250%5CnG1%20F8000%5CnG1%20X81%5CnG1%20Y273%22%7D,%22gcode_macro%20wipe_exit%22:%7B%22description%22:%22G-Code%20macro%22,%22gcode%22:%22G90%5CnG1%20F8000%5CnG1%20X83%5CnG1%20Y250%22%7D,%22gcode_macro%20wipe_nozzle%22:%7B%22description%22:%22G-Code%20macro%22,% [... cut ...] %5C%22%22:%7B%22points%22:%5B%5B-0.757167,-0.551667,-0.3605,-0.295667,-0.182167%5D,%5B-0.5725,-0.342,-0.070167,-0.076667,-0.057%5D,%5B-0.491833,-0.2045,0.0285,-0.0115,-0.110667%5D,%5B-0.593333,-0.406667,-0.180333,-0.256,-0.413167%5D,%5B-0.770833,-0.615667,-0.4625,-0.467,-0.733833%5D%5D,%22mesh_params%22:%7B%22min_x%22:5.0,%22max_x%22:245.0,%22min_y%22:5.0,%22max_y%22:245.0,%22x_count%22:5,%22y_count%22:5,%22mesh_x_pps%22:2,%22mesh_y_pps%22:2,%22tension%22:0.2,%22algo%22:%22lagrange%22%7D%7D%7D `

Clon1998, 2025-04-13:
Hey,
my parsing code is based of the official Klipper code.
Currently I do not use these values and should be able to adjust them.

dJOS1475, 2025-04-13:
Thank you @Clon1998 that would be great. AnyCubic have messed with quite a few things.

Clon1998, 2025-04-13:
Can you share the JSON output of the configfile request, just to make sure its just the microstepping field?

`GET http://YOUR-PRINTER-IP/printer/objects/query?configfile`

Clon1998, 2025-04-13:
> Can you share the JSON output of the configfile request, just to make sure its just the microstepping field?
> 
> `GET http://YOUR-PRINTER-IP/printer/objects/query?configfile`

Nvm. the JSON is included in the error :)

dJOS1475, 2025-04-13:
@Clon1998 lol, I was just grabbing it and saw your post..

`{"result": {"eventtime": 131091.801542505, "status": {"configfile": {"config": {"bed_mesh": {}, "bed_mesh default": {}, "buttons": {}, "controller_fan controller_fan": {}, "cs1237": {}, "display_status": {}, "engine_mode": {}, "exclude_object": {}, "extruder": {}, "fan": {}, "fan_generic air_filter_fan": {}, "fan_generic box_fan": {}, "filament_hub": {}, "filament_tracker": {}, "flow_calibration": {}, "force_move": {}, "gcode_arcs": {}, "gcode_macro CANCEL_PRINT": {}, "gcode_macro WIPE_ENTER": {}, "gcode_macro WIPE_EXIT": {}, "gcode_macro WIPE_NOZZLE": {}, "gcode_macro WIPE_NOZZLE1": {}, "gcode_macro WIPE_STOP": {}, "gcode_macro WIPE_STOP1": {}, "heater_bed": {}, "heater_fan extruder_fan": {}, "input_shaper": {}, "led_pin": {}, "leviQ3": {}, "lis2dw12": {}, "mcu": {}, "mcu nozzle_mcu": {}, "ota_filament_hub": {}, "pause_resume": {}, "pid_calibrate": {}, "printer": {}, "printer_marco": {}, "probe": {}, "resonance_tester": {}, "safe_z_home": {}, "stepper_enable": {}, "stepper_x": {}, "stepper_y": {}, "stepper_z": {}, "timelapse": {}, "tmc2209 extruder": {}, "tmc2209 stepper_x": {}, "tmc2209 stepper_y": {}, "tmc2209 stepper_z": {}, "verify_heater extruder": {}, "verify_heater heater_bed": {}, "virtual_sdcard": {}, "gcode_macro pause": {}, "gcode_macro resume": {}, "gcode_macro cancel_print": {}}, "save_config_pending": true, "save_config_pending_items": {"bed_mesh default": {"algo": "lagrange", "max_x": "245", "max_y": "245", "mesh_x_pps": "2", "mesh_y_pps": "2", "min_x": "5", "min_y": "5", "points": "-0.804667, -0.584000, -0.404333, -0.338833, -0.210500\n-0.624833, -0.400000, -0.106000, -0.100833, -0.075167\n-0.526500, -0.240667, -0.010000, -0.040667, -0.099667\n-0.616667, -0.402667, -0.184833, -0.242333, -0.387833\n-0.763000, -0.599500, -0.452500, -0.494500, -0.697000", "tension": "0.2", "version": "1", "x_count": "5", "y_count": "5"}, "leviQ3": {"auto_zoffset_on_off": "1"}, "probe": {"z_offset": "-0.0775"}}, "settings": {"bed_mesh": {"algorithm": "lagrange", "bicubic_tension": 0.2, "fade_end": 0, "fade_start": 1, "fade_target": 0, "horizontal_move_z": 3, "mesh_radius": 0, "move_check_distance": 5, "speed": 300, "split_delta_z": 0.0125}, "bed_mesh default": {"algo": "lagrange", "max_x": 245, "max_y": 245, "min_x": 5, "min_y": 5, "tension": 0.2}, "controller_fan controller_fan": {"cycle_time": 0.01, "fan_speed": 1, "heater": ["extruder"], "idle_speed": 0, "kick_start_time": 0.5, "max_power": 1, "off_below": 0, "pin": "PB5", "shutdown_speed": 0, "stepper": []}, "cs1237": {"dout_pin": "nozzle_mcu:PA6", "level_pin": "nozzle_mcu:PA7", "sclk_pin": "nozzle_mcu:PA5"}, "extruder": {"control": "pid", "dir_pin": "!nozzle_mcu:PA11", "enable_pin": "!nozzle_mcu:PC13", "filament_diameter": 1.75, "gear_ratio": [], "heater_pin": "nozzle_mcu:PB8", "inline_resistor": 0, "instantaneous_corner_velocity": 1, "max_extrude_cross_section": 50, "max_extrude_only_accel": 5321.6216073828855, "max_extrude_only_distance": 100, "max_extrude_only_velocity": 159.64864822148658, "max_power": 1, "max_safe_temp": 320, "max_temp": 340, "min_extrude_temp": 170, "min_safe_temp": 170, "min_temp": -180, "nozzle_diameter": 0.4, "nozzle_material": "brass", "pid_kd": 108.268, "pid_ki": 0.698, "pid_kp": 17.393, "pressure_advance": 0.03, "pressure_advance_max": 0.1, "pressure_advance_smooth_time": 0.04, "pullup_resistor": 4700, "pwm_cycle_time": 0.1, "rotation_distance": 6.5756, "sensor_pin": "nozzle_mcu:PA0", "sensor_type": "NTC ZM3486 HNS112HC4570G0060Y3486", "smooth_time": 1, "step_pin": "nozzle_mcu:PA12"}, "fan": {"cycle_time": 0.005, "kick_start_time": 0.5, "max_power": 1, "off_below": 0, "pin": "nozzle_mcu:PB6", "shutdown_speed": 0}, "fan_generic air_filter_fan": {"cycle_time": 3.333333e-05, "kick_start_time": 0.5, "max_power": 1, "off_below": 0, "pin": "PC7", "shutdown_speed": 0}, "fan_generic box_fan": {"cycle_time": 3.333333e-05, "kick_start_time": 0.5, "max_power": 1, "off_below": 0, "pin": "PC6", "shutdown_speed": 0}, "filament_hub": {"action_timeout": 240, "config_file": "/userdata/app/gk/config/ams_config.cfg", "cruising_speed": 200, "cutter_pin": "nozzle_mcu:PB1", "cutter_position": 1.5, "default_feed_speed": 30, "default_unwind_speed": 20, "delay_before_throw": 3, "feed_max_length": 2600, "flush_multiplier": 1.5, "impact_position_x": 260, "impact_position_y": 25, "impact_speed": 10, "pre_sweep_position_x": 47, "pre_sweep_position_y": 230, "recalibrate_period": 7200, "serial": "", "serial1": "", "serial_prefix": "usb-ANYCUBIC_ACE", "shake_filament_speed": 100, "sweep_fan_speed": 0, "sweep_position_x": 47, "sweep_position_y": 276, "sweep_reprint_extrude_len": 0.9, "sweep_retract_len": 0.4, "sweep_speed": 10, "switch_pin": "nozzle_mcu:PB0", "throw_impact_speed": 200, "throw_position_x": 47, "throw_position_y": 276, "unwind_length_after_triggered": 1100, "unwind_speed_old_ace": 15}, "filament_tracker": {"length_per_pulse": 1.86532063806894, "safe_unwind_len": 150, "signal_type": "adc", "tracker_detect_pin": "PB0", "tracker_encoder_pin": "PB1"}, "flow_calibration": {"delta_press_max": 60000, "fit_val_max": 20000}, "gcode_arcs": {"resolution": 1}, "gcode_macro cancel_print": {"description": "G-Code macro", "gcode": "CANCEL_PRINT_BASE", "rename_existing": "CANCEL_PRINT_BASE"}, "gcode_macro wipe_enter": {"description": "G-Code macro", "gcode": "G90\nG1 F36000\nG1 Y250\nG1 F8000\nG1 X81\nG1 Y273"}, "gcode_macro wipe_exit": {"description": "G-Code macro", "gcode": "G90\nG1 F8000\ [... cut ...] lse, "hold_current": 0.8, "home_speed_threshold": 5000, "irun_home": 13, "run_current": 1.6, "sense_resistor": 0.075, "stealthchop_threshold": 0, "switch_addr": 0, "uart_pin": "PA3"}, "verify_heater extruder": {"check_gain_time": 30, "heating_gain": 1, "hysteresis": 5, "max_error": 120}, "verify_heater heater_bed": {"check_gain_time": 300, "heating_gain": 1, "hysteresis": 5, "max_error": 120}, "virtual_sdcard": {"force_restart_when_timer_too_close": 0, "on_error_gcode": ""}}, "warnings": []}}}}`

Clon1998, 2025-04-30:
Hey,
just wanted to let you know that the latest TestFlight build on iOS and the open beta on Android are available now and include the fix.

closed 2025-05-22 by Clon1998

dJOS1475, 2025-06-07:
@Clon1998 a huge thank you for supporting the Kobra’s, it’s been working great! 🙏

[image]

commit 4fbada77 (2025-04-13): fix: Microsteps in no longer a required field.

By default klipper REQUIRES that the microsteps field is present for any stepper config. However, some custom forks do not obey to this rule. Since MR currently does not use this information I made it optional.

fixes #505