Clon1998/mobileraker #507: Cannot connect to "Kinematics: None" printers.

RubberEyes, 2025-05-02:
## Bug Report

### Description
Mobileraker throws errors when connecting to a no kinematics printer.
In addition to my regular printer instance that is able to connect with Mobileraker correctly, I also have a separate printer instance on the same computer that handles a dry box for my filaments. It occasionally throws I2C connection errors with the Temperature and Humidity sensor that I would like to be notified of via Mobileraker.

### Steps to Reproduce
1. Attempt to connect to a printer who's printer.cfg configuration has it set as `kinematics: none`
2. Be presented with the following error:
```
Exception:
 MobilerakerException{Missing field: motionReport, parentException: null, parentStack: null}

#0      PrinterBuilder.build (package:common/data/dto/machine/printer_builder.dart:165)
#1      PrinterService.refreshPrinter (package:common/service/moonraker/printer_service.dart:188)
<asynchronous suspension>
```

### Expected Behavior
It should be able to connect to a no kinematics and simply not display controls related to motion.

### Screenshots

[image]

### Version Information

- Device-OS (Android/iOS): Android
- Mobileraker: 2.8.7-618
- Klipper: v0.13.0-69-g5b2f8104
- Moonraker: v0.9.3-74-g1c124149

### Debug Logs
Please attach the log files generated by Mobileraker. You can find them by navigating to the app's settings page, scrolling to the bottom, and pressing the `Debug Logs` button.

### Additional Context
Add any other relevant context about the problem here.

---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ x ] Provided a clear bug description.
- [ x ] Listed detailed steps to reproduce the issue.
- [ x ] Described the expected behavior.
- [ x ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ x ] Specified the version numbers of Klipper and Moonraker if applicable.

RubberEyes, 2025-05-02:
I will add that notifications do seem to be working fine in spite of the inability to access the printer interface.

Clon1998, 2025-05-22:
Hey,
sorry for my late response.
I will need to adjust some of the core data structures in order to get this working.

This might take a bit more time to get done but should be possible.