Clon1998/mobileraker #509: Bed mesh view misleading with dense mesh

vvuk, 2025-05-07:
I'm doing a 50x50 mesh and the view inside Mobileraker had me thinking something was wrong, until I looked closer:

[image]

It looks like Mobileraker draws a circle splat for each point; these are getting overlaid on top of eachother, leading to the odd-looking pattern. Maybe if the mesh is more dense than some threshold, just draw squares that are 1/dim in size so that they don't overlap?

Clon1998, 2025-05-07:
Hey,
can you show me the bed mesh of fluid/mainsail?
Since I am using a plotting lib in a kinda hacky way I agree that for really dense meshes it might look off. However, for my printers it still always was a good indicator.

Clon1998, 2025-08-24:
Implemented in 
2753cce2b34e39fa67bd11cc42706c2dc3716b4f 224860f483eec548e2b8b424825e404839468c81 417a9fe73244f0e0fd55983ab50457e384556d10 213f48098c48c32ef0a8450d0f695d332c52a333

Will be included in the next release

closed 2025-08-24 by Clon1998