Clon1998/mobileraker #511: No sound on Iphone notifications

tballarin, 2025-05-20:
## Bug Report

### Description
I don't get any sound from Mobileraker notifications on my iPhone.
All notifications are enabled in my iPhone settings.

I tried disabling notifications before enabling them again.
I tried uninstalling and reinstalling the app.

### Version Information
- Device-OS (Android/iOS): iOS 18.4
- Mobileraker: 2.8.7-619

Has anyone else had this problem?

Regards

Clon1998, 2025-05-22:
Hey,
I've just tested it on my iOS devices for the custom notifications and print Status notifications and they work fine including a sound.

Are you sure the silent mode is disabled on your device?

tballarin, 2025-05-22:
Yes Iam, sound work on all my other apps