Clon1998/mobileraker #513: Unable to add webcam without errors

dJOS1475, 2025-05-23:
## Bug Report

### Description
Adding a webcam to my Kobra S1 it throws an error and won’t let me save changes.

### Steps to Reproduce
Add camera
Save camera

### Expected Behavior
I Can add a camera, but when I hit save it throws an error and won’t let me change to the Adaptive option

### Screenshots

[image]
[image]

### Version Information

- Device-OS iOS 
- Mobileraker: 2.8.8
- Klipper: GoKlipper
- Moonraker: not sure

### Debug Logs

Mipeg streamURI:
http://192.168.1.151/webcam?action=stream
Mipeg snapshotURI:
http://192.168.1.151/webcam?action=snapshot

Error Details:
MobilerakerDioException [DioExceptionType.badResponse]:
This exception was thrown because the response has a status code of 400 and RequestOptions.validateStatus was configured to throw for this status code.
The status code of 400 has the following meaning: "Client error - the request contains bad syntax or cannot be fulfilled"
Read
more
about
status
codes
at
https://
developer.mozilla.org/en-US/docs/Web/HTTP/Status

justM4D, 2025-05-23:
Because this issue was also posted in the rinkhals community I just tried it out with the full URL including port, which seems to do the trick.

So instead of the shorthand form that is default, try using http://[URL]:8080/?action=stream

dJOS1475, 2025-05-24:
There’s definitely something weird going on in the iOS app, I get various errors when attempting to save settings.

[image]

[image]

dJOS1475, 2025-05-24:
There’s definitely a bug in the iOS app, Android users are reporting it’s working for them.

I had to delete my printer and re add it, but the app still failing to save changes - currently i got it to accept the full URL, but can’t change to adaptive mjpegstreamer.

[image]

Android version:

[image]

Clon1998, 2025-05-24:
Hey,
can you please share the app's log file file with me? It is located at the very bottom of the app settings page.

dJOS1475, 2025-05-24:
> Hey, can you please share the app's log file file with me? It is located at the very bottom of the app settings page.
> 
> I recently changed to the newer webcam API (which is 2 years old by now) but some moonraker/klipper versions might be missing that.

Sure thing, here they are.

[talker_logs_2025-05-24 15 28 38.116055.txt](https://github.com/user-attachments/files/20421416/talker_logs_2025-05-24.15.28.38.116055.txt)

Clon1998, 2025-05-24:
Hey,
I am trying to understand what is going on.
In this issue you so far mentioned several problems.

1. The App is not connecting to the WebCam of your Kobra
2. The app refuses to open the machine settings page
3. The app can not save your changes in the machine settings
4. The print time duration after completing a job keeps counting up

Is that correct or did I miss something?

Clon1998, 2025-05-24:
Regarding the webcam.

Can you verify that if you open the URI: http://192.168.1.151:8080/webcam/?action=stream in your browser, you are able to see the webcam? If not, you configured the wrong absolute path/URI to the webcam. Maybe omit the port 8080 to 80 in that case?

dJOS1475, 2025-05-24:
> Hey, I am trying to understand what is going on. In this issue you so far mentioned several problems.
>
> 1. The App is not connecting to the WebCam of your Kobra
> 2. The app refuses to open the machine [...]

Howdy, pretty much. 

Regarding point 2. Frequently after I hit save and then MR tells it can’t can’t connect to the webcam, then if I go into printer settings it crashes to a white screen.

Other times it’ll throw an error saying it can’t save the settings. When it does save the settings it’s very random, most times it appears to, but doesn’t.

dJOS1475, 2025-05-24:
> Regarding the webcam.
> 
> Can you verify that if you open the URI: http://192.168.1.151:8080/webcam/?action=stream in your browser, you are able to see the webcam? If not, you configured the wrong absolute path/URI to the webcam. Maybe omit the port 8080 to 80 in that case?

I took webcam out and now it works, took me a few goes to get it stick, I had to delete the webcam and re add a couple of times.

Strangely, I can’t get the app to crash to a white screen anymore, not that I’m complaining. 😎

[image]
[image]

dJOS1475, 2025-05-24:
> [@dJOS1475](https://github.com/dJOS1475) leave the "webcam" part out of the URL.
> 
> [@Clon1998](https://github.com/Clon1998) no 4 is a kobra issue and not your app's fault.

Thanks mate, that worked.

dJOS1475, 2025-05-25:
There is some sort of app bug still present, if I select the adaptive mpegstreamer option and save, it doesn’t stick, but it doesn’t throw an error either.

Clon1998, 2025-06-05:
> Having the same problem on Android and for some reason to me it seems as if a "ghost camera" is created with the correct data, but not activated/listed.
> 
> When I changed the name from "Default" to something I recognize I saw that in the printer setup page the camera is not listed after hitting save. There are also no errors shown and preview worked. Exiting from the details page and back to the main page the preview loading fails, but clicking through to the actual printer shows the camera menu now having 2 cameras available - one named "Default" which does not work and one with "myname" which works.
> 
> Also, whenever I wrote IP:Port/webcam/?action=xxx the error message said IP:Port/webcam?action?xxx omitting the vital "/" - no clue if stripping that out only happens in the message or also the actual code.

Can you provide me your app logs alongside your moonraker and app versions?

avxmw, 2025-06-05:
Logfiles (used "copy all" and pasted into file): [logs.txt](https://github.com/user-attachments/files/20617440/logs.txt)

Note, these might not be complete as I cleared the apps Android cache earlier hoping that would help.

App is version 2.9.0-641 according to the bottom of the settings page which according to my Playstore updates page is current (beta enabled).

Moonraker only reports version as "?" both on IP:7125 and in log, don't know if that is listed anywhere else I can get access with SSH to.

Machine affected is a "Kobra 3 Combo" on firmware 2.4.04 plus Rinkhals 20250531_01, I did not try changing the configuration for my Creality K1 in mobileraker to see if it happens there too, if absolutely need be, let me know. Not tonight though, gut's nächtle.

Clon1998, 2025-06-06:
> Logfiles (used "copy all" and pasted into file): [logs.txt](https://github.com/user-attachments/files/20617440/logs.txt)
>
> Note, these might not be complete as I cleared the apps Android cache [...]

Thanks.
The logs do not state anything in particular, can you do me a favor and force close the app. Start it again and than try to modify the webcam?

Regarding the moonraker version, you should be able to find it in the moonraker log file or by opening the printer's moonraker website. This by default is: <IP>:7125 e.g. http://192.168.178.135:7125/

[image]

avxmw, 2025-06-07:
As already said above, moonraker does not know what version it is, I guess either Anycubic (more likely) or Rinkhals tempered with it for some reason.

[image]

[image]

I cleared all data (from the androids app settings page), then uninstalled the app and reinstalled. What I found confusing is that the dashboard does not load if it can not contact one or more of googleapis/gstatic/amazonaws, from what I can see that is not even a request for ads. Clicking to the console or files works to get access to the machine though so that is annoying.

Anyhow, that aside, although I cleared data once I added the K3C again - even under a different profile name - for some reason the old cameras I configured earlier in the app showed up again, but are not shown on fluidd/mainsail on the machine itself.

Making a new camera "test" works as seen in "preview", but saving fails. Allowing internet access and creating a "test2" camera with the same data allows saving and actually works, however data also gets sent to google+amazon URLs for that.

Not sure what that means, new logs here 

[talker_logs_2025-06-07 14 25 18.641166.txt](https://github.com/user-attachments/files/20638628/talker_logs_2025-06-07.14.25.18.641166.txt)

Clon1998, 2025-06-07:
Got it! I initially thought it was a stock version of Moonraker. However, I am quite certain that the version currently in use does not support the newer webcam API, which uses UIDs instead of webcam names to reference webcam entries in the Moonraker database.

Regarding the dashboard and the Google and AWS URIs, as you've probably noticed in the codebase, I utilize Firebase (from Google) for stuff like crash tracking,  push notifications and more. I don't believe I'm using AWS for anything directly, but the third-party add-on I use for managing purchases (RevenueCat) may rely on AWS.

Could you please provide a recording or a screenshot showing what the dashboard looks like, especially after you blocked the AWS and Google URIs?

I will review the provided app logs to determine if the modified version of Moonraker in use is causing the issues with the webcam.

dJOS1475, 2025-06-07:
> Machine affected is a "Kobra 3 Combo" on firmware 2.4.04 plus Rinkhals 20250531_01, I did not try changing the configuration for my Creality K1 in mobileraker to see if it happens there too, if absolutely need be, let me know. Not tonight though, gut's nächtle.

You need to put in the full URL, removing webcam

http://192.168.1.151:8080/webcam/?action=stream

Like this

http://192.168.1.151:8080/?action=stream

Then it works fine. You might need to delete the webcam and start fresh to get around the app crashing.

[image]