Clon1998/mobileraker #518: No snapchot notification

Ren0oren, 2025-06-26:
## Bug Report

### Description
Since a few days i don't get snapchot with progress notification. 
I am a lifetime membership.
I am on Android 15

### Expected Behavior
Before i get the notification with a picture pf m'y print 

### Version Information

- Device-OS (Android/iOS): Android 15
- Mobileraker: 2.9
- Klipper: 0.12.0-639
- Moonraker: 0.5.0-2

### Debug Logs

[talker_logs_2025-06-26 11 08 33.210845.txt](https://github.com/user-attachments/files/20923071/talker_logs_2025-06-26.11.08.33.210845.txt)

### Additional Context
Add any other relevant context about the problem here.

---

Clon1998, 2025-06-26:
> Hi Yes i confirm i got notification but non pictures attached to it
> 
> [image]

Got it.
Can you they to force close and open the app again?
Maybe the sync failed. Otherwise I'll need to investigate that further during the weekend.

Ren0oren, 2025-06-26:
I restore the purchase but no change . Thé lifetime is set top Activ thé only thing is the vérification email stay but i've confirmed m'y email many time 

[image]

[image]

Ren0oren, 2025-06-26:
Thank you so mutch that's it. The webcam was disabled in the parameter. Reselect the cam and that's ok. Thanks for this wonderfull application 

[image]