Clon1998/mobileraker #519: Broken webcam access

sTiKyt, 2025-06-30:
[image]

[image]

sTiKyt, 2025-08-15:
And if camera has no such thing?

Clon1998, 2025-08-15:
> And if camera has no such thing? 

IT fails since the configuration of you cam is a plain IP without a schema. Include th Schema and it will work or add the webcam in mobileraker directly.

sTiKyt, 2025-08-15:
> > And if camera has no such thing?
> 
> IT fails since the configuration of you cam is a plain IP without a schema. Include th Schema and it will work or add the webcam in mobileraker directly.

Look, this is how it works:

[image]

[image]

There is no schema or anything like that

sTiKyt, 2025-08-15:
And there's no way for me to add it in mobileraker, because if i hit add nothing happens ,and error is shown

[image]