Clon1998/mobileraker #531: Request to support printer with multiple extruder

jas0nc, 2025-09-08:
## Feature Request
Request to support printer with multiple extruder

### Problem Description
I am using a tool changer printer (stealthchanger) with 4 toolhead/extruders. when I want to extrude/retract with mobileraker, the extruder UI card is grey out. And this endless long grey card also becomes an obstacle for me to use feature below extruder card. see screenshot for reference.

### Proposed Solution
Please add support to tool changer, so I can control my stealthchanger in the app.

### Alternatives Considered
if it is not in your priority, please at least allow me to disable certain card in the UI, so I can use other features.

### Additional Context
Screenshot is attached.
[image]

---

**Checklist**
To help us understand your feature request, please ensure you've covered the following points:
- [v] Provided a clear description of the problem you're facing.
- [v] Clearly outlined the proposed solution or feature.
- [v] Mentioned any alternative solutions or ideas you've considered.
- [v] Included any relevant context or supporting materials.

Your input is valuable in shaping the development of our project. Thank you for taking the time to submit your feature request!

Clon1998, 2025-09-13:
Hi,  
Could you please share the device logs with me? It seems my app has a parsing error when connected to your printer.  
It works fine with my IDEX Vcore 4.  
Best regards,  
Patrick

jas0nc, 2025-09-15:
Hi Patrick,

Thanks for ordering your help. please see my log file as attached.

[talker_logs_2025-09-15 11 23 59.229570.txt](https://github.com/user-attachments/files/22326782/talker_logs_2025-09-15.11.23.59.229570.txt)

jas0nc, 2025-10-07:
No problem. The wired value maybe related to spoolman. Here is the output for your reference: {"result":{"eventtime":2359938.889019806,"status":{"gcode_macro T0":{"color":"c44","spool_id":14}}}}

Clon1998, 2025-10-07:
> No problem. The wired value maybe related to spoolman. Here is the output for your reference: {"result":{"eventtime":2359938.889019806,"status":{"gcode_macro T0":{"color":"c44","spool_id":14}}}}

Thanks, that's confirming to me that the `color` variable for any of the Toolhead macros has a very weird value. Do you mind sharing a klipper log with me to have a look at the configured gcodes macros to see where it is set?

In any case, tho, I will fix that issue since this should not stop my app from displaying the card correctly...

commit 79000189 (2025-10-07): fix: Ensure we display the tool selection even if the color/colour var of the tool macro contains a non hex color.

fixes #531
closed 2025-10-07 by Clon1998