Clon1998/mobileraker #532: Can't start New Job from Overview screen if files are inside a folder

wikolii, 2025-09-09:
## Bug Report

### Description
Can't start New Job from Overview screen if files are inside a folder.
It will work from root gcode directory

### Steps to Reproduce
1. Go to 'Overview screen'
2. Click on 'New Job' on any printer
3. Scroll down to 'Any chosen folder'
4. Select file to print
5. The menu goes back to the root gcode folder
6. There is no error message or anything

### Expected Behavior
Start the job located inside a folder

### Version Information

- Device-OS (Android/iOS): iOS 18.6.2
- Mobileraker: 2.9.1-656
- Klipper: v0.13.0-213-gd34d3b05
- Moonraker: v0.9.3-102-gf20feff6

### Debug Logs

[talker_logs_2025-09-09 09 40 15.706596.txt](https://github.com/user-attachments/files/22234365/talker_logs_2025-09-09.09.40.15.706596.txt)

### Additional Context
I can start jobs from each printer interface even if inside folders.
Starting jobs from the overview screen seems to only work for files in the root gcode directory
---

#### Checklist
To help us diagnose the issue, please ensure you've completed the following steps:
- [ ] Provided a clear bug description.
- [ ] Listed detailed steps to reproduce the issue.
- [ ] Described the expected behavior.
- [ ] Included the Mobileraker version you are using.
- [ ] Attached Mobileraker's debug log files.
- [ ] Specified the version numbers of Klipper and Moonraker if applicable.

Clon1998, 2025-09-09:
Thanks for reporting the issue.
That's not how it's supposed to work.
I will have a look if I can hot fix it.

Clon1998, 2025-09-14:
Fixed in edcef74179458e298a7af9ade52cde2e0b762051

wikolii, 2025-09-15:
Hi. Tried the version available in testlight and saw that the behavior change:
Now when I start a job from a file inside a folder it goes back to the overview screen and nothing else happens.
If I open that machine nothing change and if I go back to the overview screen the app beco es gray and there nothing else i can do except closed the app conpletely to have access again.
The job still cannot be started from yhe overview screen if inside a folder

Thanks

wikolii, 2025-09-15:
Correct, all other functions work as before, as far as i could test in this short period
Thanks

closed 2025-10-05 by Clon1998