Clon1998/mobileraker #537: WebRTC - Default URL incorrect

DrFate09, 2025-09-25:
I have WebRTC setup with a Crowsnest config running within Fluidd. My host is a RPi 4B running Bookworm. 

I currently do not have my printer connected to my RPi, so Mobileraker is only connecting to the RPi. 

Mobileraker is adding a /%20/ to the URL and it is breaking the camera connection. If I go on my browser and remove the /%20/, the camera stream loads correctly. 

Can we remove the unnecessary/%20/ from the default WebRTC URL? 

[image]

DrFate09, 2025-09-25:
These are the only settings that I have available while having a MCU disconnected. 

[image]

DrFate09, 2025-10-01:
No, it was set up initially using Fluidd & Crowsnest.

Clon1998, 2025-10-01:
> No, it was set up initially using Fluidd & Crowsnest. 

In that case the space is coming from fluidd